jiangdongguo/AndroidUSBCamera #2: Error 12291 on opening app sometimes (and always when app is opened through android studio when screen is locked)

v4-adi, 2017-10-26:
java.lang.IllegalArgumentException: java.lang.RuntimeException: createWindowSurface failed error=12291
                                                                                          at com.serenegiant.glutils.EGLBase14.createWindowSurface(EGLBase14.java:466)
                                                                                          at com.serenegiant.glutils.EGLBase14.access$100(EGLBase14.java:44)
                                                                                          at com.serenegiant.glutils.EGLBase14$EglSurface.<init>(EGLBase14.java:90)
                                                                                          at com.serenegiant.glutils.EGLBase14$EglSurface.<init>(EGLBase14.java:79)
                                                                                          at com.serenegiant.glutils.EGLBase14.createFromSurface(EGLBase14.java:192)
                                                                                          at com.serenegiant.glutils.EGLBase14.createFromSurface(EGLBase14.java:43)
                                                                                          at com.serenegiant.usb.widget.UVCCameraTextureView$RenderHandler$RenderThread.init(UVCCameraTextureView.java:564)
                                                                                          at com.serenegiant.usb.widget.UVCCameraTextureView$RenderHandler$RenderThread.run(UVCCameraTextureView.java:543)
                                                                                       Caused by: java.lang.RuntimeException: createWindowSurface failed error=12291
                                                                                          at com.serenegiant.glutils.EGLBase14.createWindowSurface(EGLBase14.java:460)
                                                                                          at com.serenegiant.glutils.EGLBase14.access$100(EGLBase14.java:44) 
                                                                                          at com.serenegiant.glutils.EGLBase14$EglSurface.<init>(EGLBase14.java:90) 
                                                                                          at com.serenegiant.glutils.EGLBase14$EglSurface.<init>(EGLBase14.java:79) 
                                                                                          at com.serenegiant.glutils.EGLBase14.createFromSurface(EGLBase14.java:192) 
                                                                                          at com.serenegiant.glutils.EGLBase14.createFromSurface(EGLBase14.java:43) 
                                                                                          at com.serenegiant.usb.widget.UVCCameraTextureView$RenderHandler$RenderThread.init(UVCCameraTextureView.java:564) 
                                                                                          at com.serenegiant.usb.widget.UVCCameraTextureView$RenderHandler$RenderThread.run(UVCCameraTextureView.java:543)

805447391, 2018-03-08:
I have same problem.

805447391, 2018-03-08:
I guess that maybe happened when UVCCameraTextureView is creating. but some activity over it's owner activity, so GL can not init well.
Can everyone tell me the bast way to deal with this problem?