jiangdongguo/AndroidUSBCamera #4: 大佬这库是不是和opencv的库有冲突？

as8190255, 2017-11-25:
大佬您好，我在单独用您的库后非常顺利，然后整合进我的项目后，就会遇到OpenCV library not found!。接下来不知道该怎么办了。~~~~(>_<)~~~~

jiangdongguo, 2017-11-27:
这个我没有遇到过。你试试 将我的库移除，然后会不会报，再添加我的库，再看会不会报，如果会就说明有影响

as8190255, 2017-11-27:
感谢大佬回答。是的，什么代码都不加的情况下一导入后就会出现opencv无法加载，估计是底层库冲突了。最后解决方法只好使用原作者的另一份代码，这里面已经包含了uvc摄像头和opencv的代码了。https://github.com/saki4510t/OpenCVwithUVC

closed 2017-11-27 by as8190255

as8190255, 2017-11-27:
大佬再问你个事情，你有没有注意到，在使用摄像头的时候，如果程序出现了异常，或者这个时候在开发工具上再run一次程序。那么接下来app就不会显示图像了，只能重启手机或者重插拔usb后才会显示正常。 运行后日志显示
E/libuvc/stream: [11522*stream.c:1605:uvc_stream_start_bandwidth]:fail
W/libUVCCamera: [11522*diag.c:88:uvc_perror]:failed start_streaming:Not found (-5)
感觉好像是强制退出后导致底层代码没有释放摄像头才会这样

momxmo, 2021-06-23:
> 大佬再问你个事情，你有没有注意到，在使用摄像头的时候，如果程序出现了异常，或者这个时候在开发工具上再run一次程序。那么接下来app就不会显示图像了，只能重启手机或者重插拔usb后才会显示正常。 运行后日志显示
> E/libuvc/stream: [11522_stream.c:1605:uvc_stream_start_bandwidth]:fail W/libUVCCamera: [11522_diag.c:88:uvc_perror]:failed start_streaming:Not found (-5)
> 感觉好像是强制退出后导致底层代码没有释放摄像头才会这样

这个问题你解决了吗？