jiangdongguo/AndroidUSBCamera #7: 4.4.4平板无法正常使用

Haskell2015, 2017-12-16:
使用环境：4.4.4系统的平板
bug描述：clone运行之后，打开app之后首先无法预览，而后过一会就会闪退
报错信息：
`12-16 10:52:03.649 4180-4229/com.jiangdg.usbcamera A/libc: Fatal signal 11 (SIGSEGV) at 0x6efdd000 (code=2), thread 4229 (CameraThread)`

jiangdongguo, 2017-12-27:
换个版本高的平板试试   4.4.4太低了

Haskell2015, 2017-12-28:
嗯，后来发现不需要通过JNI，直接自定义camera处理了，感谢大佬的开源

closed 2017-12-28 by Haskell2015

UbuntuLover, 2018-01-03:
握草 你不是在[https://github.com/a-voyager/AutoInstaller](url)这里也问问题的那个人吗 。。。

Haskell2015, 2018-01-03:
楼上的老哥，是我==缘分啊 @UbuntuLover

gxh-apologize, 2018-06-15:
你好，这个库挺好，但是在某个开发板上遇到了跟你一样的问题，所以换成了Camera2预览，同样的代码，在手机、平板运行良好（不是预览的usb摄像头），但是在开发板上看usb摄像头，预览的方向始终是竖屏的（我的activity是横屏的），而且拉伸变形。从开发板自带相机看，也是拉伸竖屏。。这种情况下如何处理呢？刚接触Camera2，希望回复下，谢谢。

UbuntuLover, 2018-06-15:
@gxh-apologize 你可以自己写一个view来继承TextureView，然后onMeasure方法里面来测量尺寸然后跟Size进行选择匹配。