jiangdongguo/AndroidUSBCamera #12: 第一次进去正常，退出再进黑屏

thearyong, 2018-01-02:
1.集成在自己的app 中，第一次进去正常打开。
2.退出当前页面后，再进黑屏。无报错。log 提示：
`E/BufferQueueProducer: [SurfaceTexture-1-25464-3](this:0x7dee11a800,id:3,api:2,p:25464,c:-1) dequeueBuffer: BufferQueue has been abandoned`
3.杀死 app再进，又可以打开。
4.测试机是 android7.0开发板，MTK6737平台。
怀疑是资源释放问题，但是有调用release的。

jiangdongguo, 2018-01-05:
资源未完全释放，是否有参照我的Demo来写？onSurfaceCreated和onSurfaceDestoryed那部分

UbuntuLover, 2018-01-10:
@thearyong 我也是在6737的平台上开发。你们可以支持UVC摄像头1280*720的分辨率吗

thearyong, 2018-01-25:
@UbuntuLover 抱歉，好久没上。
也不行，当时修改最大支持640*360，不知啥问题。后来放弃此方案

UbuntuLover, 2018-01-29:
@thearyong 没事。感觉是速度的限制？USB口的速度限制还是6737芯片编解码处理的速度限制？有没有试过可以两路h264流的编解码呢

thearyong, 2018-01-31:
@UbuntuLover 可以肯定不是 USB速度限制，USB2.0 速度至少都是10+MB/s（理论都是60+）。你说的两路编解码没研究过，不是特别懂哈

UbuntuLover, 2018-02-01:
@thearyong 这块感觉可以深入研究一下。我也是在方案公司做这个。。。

jiangdongguo, 2018-04-04:
已支持高分辨率，画面流畅，请更新到2.1.0

cizkey, 2018-06-07:
@thearyong 请问兄弟是否有解决的办法?

thearyong, 2018-06-10:
@cizkey 抱歉，没有找到原因解哦，后面我们直接接入https://github.com/saki4510t/UVCCamera

cizkey, 2018-06-12:
@thearyong 我昨天直接跑UVCCamera的demo, 发现也是这种状况, 看来不是这个库的问题, 是UVCCamera库本身存在bug. 不过具体原因还没有跟出来.

bingyanxueling, 2018-06-14:
我也遇见同样的问题，UVCCamera的demo也是同样的问题，有解决方案吗？