jiangdongguo/AndroidUSBCamera #14: 下载的demo运行起来黑屏

453514644, 2018-02-25:
下载demo正常运行，但是摄像头没打开，显示黑屏，麻烦帮忙看下，453514644，如果方便能加下我qq，这块想请教下，非常感谢

453514644, 2018-02-25:
E/UVCCamera: [12775*UVCCamera.cpp:178:connect]:could not open camera:err=-99

hjhcs121, 2019-09-04:
不知解决该问题了吗？

Ruiluo123, 2020-05-28:
这个问题解决了吗？我的报这个错误。。。 
D/UsbDeviceConnectionJNI( 5411): close
 I/libusb/core( 5411): [18411*core.c:2085:libusb_init2]:libusb v1.0.19.10903
 I/libusb/usbfs( 5411): [18411*android_usbfs.c:512:op_init2]:call android_scan_devices
 E/libusb/usbfs( 5411): [18411*android_usbfs.c:1776:release_interface]:release interface failed, error -1 errno 22
E/UVCCamera( 5411): [18411*UVCCamera.cpp:178:connect]:could not open camera:err=-99
 W/System.err( 5411): java.lang.UnsupportedOperationException: open failed:result=-99----->id_camera=1871661592;venderId=3141;productId=25408;fileDescriptor=379;busNum=1;devAddr=7;usbfs=/dev/bus/usb
 W/System.err( 5411): Exception：调用nativeConnect返回值：-99
W/System.err( 5411): 	at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:217)
 W/System.err( 5411): 	at android.os.Handler.handleCallback(Handler.java:733)
 W/System.err( 5411): 	at android.os.Handler.dispatchMessage(Handler.java:95)
 W/System.err( 5411): 	at android.os.Looper.loop(Looper.java:136)
 W/System.err( 5411): 	at android.os.HandlerThread.run(HandlerThread.java:61)
 D/UsbDeviceConnectionJNI( 5411): close