jiangdongguo/AndroidUSBCamera #19: 怎么推rtmp视频流？

wasdzy111, 2018-03-13:


wasdzy111, 2018-03-14:
或者说 一点录制的时候 就抛异常了

jiangdongguo, 2018-03-14:
这里只提供h264数据流，rtmp那块需自己弄

wasdzy111, 2018-03-15:
执行录制  获得数据流的时候  就报错了   
---------------------------------------------------------------
mCameraHelper.startRecording(params, new AbstractUVCCameraHandler.OnEncodeResultListener() {
                        @Override
                        public void onEncodeResult(byte[] data, int offset, int length, long timestamp, int type) {
                            // type = 1,h264 video stream
                            if (type == 1) {
//                                FileUtils.putFileStream(data, offset, length);
                            }
                            // type = 0,aac audio stream
                            if(type == 0) {

                            }
                        }

                        @Override
                        public void onRecordResult(String videoPath) {
                            Log.i(TAG,"videoPath = "+videoPath);
                        }
                    });