jiangdongguo/AndroidUSBCamera #25: Android 4.4.4 上预览出错了，请帮忙看下

wuzhenzhen, 2018-03-29:
03-30 00:39:55.620 3216-3216/com.jiangdg.usbcamera V/USBMonitor: USBMonitor:Constructor
03-30 00:39:55.620 3216-3216/com.jiangdg.usbcamera V/USBMonitor: USBMonitor:mUsbManager=android.hardware.usb.UsbManager@41a6feb0
03-30 00:39:55.630 3216-3216/com.jiangdg.usbcamera V/CameraThread: getHandler:
03-30 00:39:55.630 3216-3216/com.jiangdg.usbcamera I/USBMonitor: register:
03-30 00:39:55.660 3216-3216/com.jiangdg.usbcamera D/dalvikvm: GC_FOR_ALLOC freed 269K, 11% free 3032K/3372K, paused 13ms, total 13ms
03-30 00:39:55.800 3216-3216/com.jiangdg.usbcamera D/OpenGLRenderer: Enabling debug mode 0
03-30 00:39:55.850 3216-3216/com.jiangdg.usbcamera I/UVCCameraTextureView: onSurfaceTextureAvailable:android.graphics.SurfaceTexture@41a35f48
03-30 00:39:55.850 3216-3216/com.jiangdg.usbcamera V/UVCCameraTextureView: RenderThread#getHandler:
03-30 00:39:55.850 3216-3231/com.jiangdg.usbcamera D/UVCCameraTextureView: RenderThread started
03-30 00:39:55.850 3216-3231/com.jiangdg.usbcamera V/UVCCameraTextureView: RenderThread#init:
03-30 00:39:55.860 3216-3231/com.jiangdg.usbcamera D/EGLBase14: EGLContext created, client version 2
03-30 00:39:55.880 3216-3216/com.jiangdg.usbcamera V/UVCCameraTextureView: getSurface:hasSurface=true
03-30 00:39:55.880 3216-3216/com.jiangdg.usbcamera V/UVCCameraTextureView: getPreviewTexture:
03-30 00:39:55.880 3216-3231/com.jiangdg.usbcamera I/UVCCameraTextureView: RenderThread#updatePreviewSurface:
03-30 00:39:55.880 3216-3231/com.jiangdg.usbcamera V/UVCCameraTextureView: updatePreviewSurface:tex_id=70001
03-30 00:39:56.680 3216-3229/com.jiangdg.usbcamera V/USBMonitor: processConnect:device=UsbDevice[mName=/dev/bus/usb/001/005,mVendorId=1423,mProductId=14464,mClass=239,mSubclass=2,mProtocol=1,mInterfaces=[Landroid.os.Parcelable;@41a4ef38]
03-30 00:39:56.690 3216-3229/com.jiangdg.usbcamera I/dalvikvm: Could not find method android.hardware.usb.UsbInterface.getAlternateSetting, referenced from method com.serenegiant.usb.USBMonitor$UsbControlBlock.getInterface
03-30 00:39:56.690 3216-3229/com.jiangdg.usbcamera W/dalvikvm: VFY: unable to resolve virtual method 1003: Landroid/hardware/usb/UsbInterface;.getAlternateSetting ()I
03-30 00:39:56.690 3216-3229/com.jiangdg.usbcamera D/dalvikvm: VFY: replacing opcode 0x6e at 0x0039
03-30 00:39:56.690 3216-3229/com.jiangdg.usbcamera I/USBMonitor: UsbControlBlock:constructor
03-30 00:39:56.710 3216-3229/com.jiangdg.usbcamera D/UsbDeviceConnectionJNI: close
03-30 00:39:56.710 3216-3229/com.jiangdg.usbcamera I/USBMonitor: name=/dev/bus/usb/001/005,desc=66,busnum=1,devnum=5,rawDesc=[B@41a28228
03-30 00:39:56.720 3216-3229/com.jiangdg.usbcamera V/UVCCameraTextureView: getPreviewTexture:
03-30 00:39:56.720 3216-3231/com.jiangdg.usbcamera I/UVCCameraTextureView: RenderThread#updatePreviewSurface:
03-30 00:39:56.720 3216-3231/com.jiangdg.usbcamera D/UVCCameraTextureView: updatePreviewSurface:release mPreviewSurface
03-30 00:39:56.720 3216-3231/com.jiangdg.usbcamera V/UVCCameraTextureView: updatePreviewSurface:tex_id=140002
03-30 00:39:56.750 3216-3230/com.jiangdg.usbcamera V/CameraThread: handleOpen:
03-30 00:39:56.750 3216-3230/com.jiangdg.usbcamera V/CameraThread: handleClose:
03-30 00:39:56.750 3216-3230/com.jiangdg.usbcamera V/UVCCameraTextureView: setVideoEncoder:
03-30 00:39:56.750 3216-3230/com.jiangdg.usbcamera D/dalvikvm: Trying to load lib /data/app-lib/com.jiangdg.usbcamera-2/libjpeg-turbo1500.so 0x41a05420
03-30 00:39:56.750 3216-3231/com.jiangdg.usbcamera V/UVCCameraTextureView: RenderThread#setEncoder:encoder=null
03-30 00:39:56.760 3216-3230/com.jiangdg.usbcamera D/dalvikvm: Added shared lib /data/app-lib/com.jiangdg.usbcamera-2/libjpeg-turbo1500.so 0x41a05420
03-30 00:39:56.760 3216-3230/com.jiangdg.usbcamera D/dalvikvm: No JNI_OnLoad found in /data/app-lib/com.jiangdg.usbcamera-2/libjpeg-turbo1500.so 0x41a05420, skipping init
03-30 00:39:56.770 3216-3230/com.jiangdg.usbcamera D/dalvikvm: Trying to load lib /data/app-lib/com.jiangdg.usbcamera-2/libusb100.so 0x41a05420
03-30 00:39:56.770 3216-3230/com.jiangdg.usbcamera D/dalvikvm: Added shared lib /data/app-lib/com.jiangdg.usbcamera-2/libusb100.so 0x41a05420
03-30 00:39:56.770 3216-3230/com.jiangdg.usbcamera D/dalvikvm: No JNI_OnLoad found in /data/app-lib/com.jiangdg.usbcamera-2/libusb100.so 0x41a05420, skipping init
03-30 00:39:56.770 3216-3230/com.jiangdg.usbcamera D/dalvikvm: Trying to load lib /data/app-lib/com.jiangdg.usbcamera-2/libuvc.so 0x41a05420
03-30 00:39:56.770 3216-3230/com.jiangdg.usbcamera D/dalvikvm: Added shared lib /data/app-lib/com.jiangdg.usbcamera-2/libuvc.so 0x41a05420
03-30 00:39:56.770 3216-3230/com.jiangdg.usbcamera D/dalvikvm: No JNI_OnLoad found in /data/app-lib/com.jiangdg.usbcamera-2/libuvc.so 0x41a05420, skipping init
03-30 00:39:56.780 3216-3230/com.jiangdg.usbcamera D/dalvikvm: Trying to load lib /data/app-lib/com.jiangdg.usbcamera-2/libUVCCamera.so 0x41a05420
03-30 00:39:56.780 3216-3230/com.jiangdg.usbcamera D/dalvikvm: Added shared lib /data/app-lib/com.jiangdg.usbcamera-2/libUVCCamera.so 0x41a05420
03-30 00:39:56.800 3216-3230/com.jiangdg.usbcamera D/UsbDeviceConnectionJNI: close
03-30 00:39:56.800 3216-3230/com.jiangdg.usbcamera I/libusb/core: [3230*core.c:2085:libusb_init2]:libusb v1.0.19.10903
03-30 00:39:56.800 3216-3230/com.jiangdg.usbcamera I/libusb/usbfs: [3230*android_usbfs.c:504:op_init2]:start up hotplug event handler
03-30 00:39:56.800 3216-3230/com.jiangdg.usbcamera I/libusb/usbfs: [3230*android_usbfs.c:512:op_init2]:call android_scan_devices
03-30 00:39:56.800 3216-3230/com.jiangdg.usbcamera I/AbsUVCCameraHandler: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x480","352x288","320x240","176x144","160x120"]}]}
03-30 00:39:56.800 3216-3230/com.jiangdg.usbcamera V/CameraThread: handleStartPreview:
03-30 00:39:56.800 3216-3230/com.jiangdg.usbcamera I/libUVCCamera: [3230*UVCPreview.cpp:276:callbackPixelFormatChanged]:PIXEL_FORMAT_YUV20SP:
03-30 00:39:56.820 3216-3234/com.jiangdg.usbcamera I/libUVCCamera: [3234*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@YUYV
03-30 00:39:56.820 3216-3236/com.jiangdg.usbcamera I/libUVCCamera: [3236*UVCPreview.cpp:276:callbackPixelFormatChanged]:PIXEL_FORMAT_YUV20SP:
03-30 00:40:04.080 3216-3233/com.jiangdg.usbcamera A/libc: Fatal signal 11 (SIGSEGV) at 0x6f0e3000 (code=2), thread 3233 (CameraThread)

jiangdongguo, 2018-03-30:
是否开启硬件加速？

wuzhenzhen, 2018-04-01:
谢谢回复，

设置里没有看到 “硬件加速” 选项，
AndroidManifest.xml 里也没有指定 android:hardwareAccelerated
是需要开启硬件加速 ？

gxh-apologize, 2018-06-02:
请问一下解决了吗这个问题？我也遇到了类似的问题

hexin9456, 2019-08-13:
这个库 支持4.4.4吗 用了 大佬 上个版本的 4.4.4 没有问题 这个版本出现了4.4.4无法预览的问题