jiangdongguo/AndroidUSBCamera #34: 反复几次打开摄像头时无预览画面

RiseMrLi, 2018-04-24:
首先感谢大神的分享. 最近使用您的库时发现, 在反复几次打开摄像头时 Toast提示Connecting后, 无预览画面出现,需要重新插拔USB后才能恢复.希望大神帮忙解答下,谢谢!

UbuntuLover, 2018-05-09:
附上日志会好一点。

wangkang6, 2018-05-11:
@RiseMrLi 我在MTK平台也遇见了这样的问题，你解决了吗

RiseMrLi, 2018-05-11:
@wangkang6 暂时还没有解决，不过我在https://github.com/saki4510t/UVCCamera/issues/350这界面里面有人给出了解答，你看下能不能帮到你

UbuntuLover, 2018-05-11:
大神们对于提高帧率有什么好的建议吗

ZhouHoubin, 2018-05-15:
兄弟解决了没有,我发现在手机上一点问题没有,平板上返回再进去就有问题了

wangkang6, 2018-05-15:
我发现每次退出activity的时候结束整个进程彻底释放资源，是不会出现上述问题的。但如果你要和其他app进行交互，这样做显然是不合理的。一直没有找到彻底释放资源的方法，所以现在还没有解决这个问题的完美方法

UbuntuLover, 2018-05-18:
其实可以封装成service，然后暴露出来调用。

cizkey, 2018-06-07:
@UbuntuLover 兄弟您是否解决了这个问题呢?

UbuntuLover, 2018-06-12:
@cizkey 抱歉 没有呢。

UbuntuLover, 2018-06-16:
@amusiaHzr 是的。你们要找硬件确认这个问题。我们也是通过修改电路之后解决了供电问题

cizkey, 2018-07-09:
确实如上面几位兄弟所说是硬件的问题. 不过我这设备通过外接电源没有解决, 是设备重新烧录了厂商优化过的新的固件解决这个问题的. 总的来说, 都是硬件问题.

UbuntuLover, 2018-07-10:
大神们对于提高帧率 ，加快IFrameCallback有什么好的建议吗

UbuntuLover, 2018-08-18:
大兄弟们这个问问题有解决方案吗？还是原生库的bug？ @RiseMrLi @wangkang6 @ZhouHoubin @cizkey @amusiaHzr

Jasonchenlijian, 2019-03-05:
@UbuntuLover 请问提高帧率后来又解决方案了吗？

LaoTie0815, 2020-03-14:
@RiseMrLi 大哥有解决不，我这边也是必须插拔重启设备，但是Windows上面没有这个问题，所以说是硬件问题他们还不认，咋办