jiangdongguo/AndroidUSBCamera #41: 关于打开app自动适配到最大分辨率

SymphonyZhang, 2018-05-17:
现在打开摄像头默认分辨率为640x480,怎样才能让app打开时就适配到最大分辨率？之后随用户要求自己更改分辨率。现在所有配置都在onResume()之前做好了，把List<Size> list = mCameraHelper.getSupportedPreviewSizes();这一句放到onResume()中，获取的list为null。能给点思路吗？这句要放哪里才会获取到摄像头所支持的分辨率呢？最好是在还没预览前。

jiangdongguo, 2018-05-19:
我提供了一个void setDefaultPreviewSize(int defaultWidth,int defaultHeight)接口  在InitMonitor之前设置即可

SymphonyZhang, 2018-05-21:
这个我知道，也用过，确实可以提前设置分辨率，但是如果我设置了一个1080p的分辨率是可以用1080p的摄像头，但是这是我要是插入一个720p的摄像头不久不行了么？所以我是想在显示前先获取支持的分辨率，然后拿到当前摄像头所支持的最大分辨率设置进去，这样就不用每次换摄像头都要改一次支持的最大分辨率然后编译一个版本的app了。

jiangdongguo, 2018-05-21:
在onConnected方法中调用mCameraHelper.getSupportedPreviewSizes(); ，然后将最大值设置给updateResolution

ibenben, 2018-06-22:
`mUSBManager.updateResolution( 800, 600, new USBCameraManager.OnPreviewListener() {
            @Override
            public void onPreviewResult(boolean isSuccess) {
                if(! isSuccess) {
                    showShortMsg("预览失败，不支持该分辨率");
                }else {
                    showShortMsg("以切换到分辨率为1280x720");
                }
            }
        });`

`implementation('com.github.jiangdongguo:AndroidUSBCamera:1.3.8') { force = true }`

设置分辨率，onPreviewResult没进去。两个showShortMsg都没触发。app界面显示全白。  不加这个设置，能显示内容。

SymphonyZhang, 2018-06-22:
我用的是implementation 'com.github.jiangdongguo:AndroidUSBCamera:2.1.0'这个版本，已经不是这种切换分辨率方式了。mCameraHelper.updateResolution(widht, height); 这里的widht 和 height都是读摄像头支持的分辨率设置进去的。

wesoho, 2019-03-08:
> 在onConnected方法中调用mCameraHelper.getSupportedPreviewSizes(); ，然后将最大值设置给updateResolution

获取到分辨率以后，使用updateResolution方法，然后app闪退了