jiangdongguo/AndroidUSBCamera #42: 调用录像的startRecording会崩溃

SymphonyZhang, 2018-05-17:
调用startRecording会崩溃，录不了像，log：
05-17 19:03:34.425 18976-19021/com.byteflyer.mcamera E/H264EncodeConsumer: couldn't find a good color format for OMX.MTK.VIDEO.ENCODER.AVC / video/avc
    Unable to find an appropriate codec for video/avc
05-17 19:03:34.626 18976-19021/com.byteflyer.mcamera E/AndroidRuntime: FATAL EXCEPTION: Thread-1516
    Process: com.byteflyer.mcamera, PID: 18976
    java.lang.NullPointerException: Attempt to invoke virtual method 'int android.media.MediaCodec.dequeueOutputBuffer(android.media.MediaCodec$BufferInfo, long)' on a null object reference
        at com.serenegiant.usb.encoder.biz.H264EncodeConsumer.run(H264EncodeConsumer.java:153)

这是解码器有问题吗？还是要设置解码？根据README的做法来的。

jiangdongguo, 2018-05-19:
编码器的color format不匹配，下载源码自己设置color format

SymphonyZhang, 2018-07-28:
我直接接收aac类型的就可以录像了。ty

closed 2018-07-28 by SymphonyZhang

fanzhouyou, 2020-08-06:
调用startRecording会崩溃，录不了像，log：
android.media.MediaCodec$CodecException: Error 0xfffffc0e
        at android.media.MediaCodec.native_configure(Native Method)
        at android.media.MediaCodec.configure(MediaCodec.java:2023)
        at android.media.MediaCodec.configure(MediaCodec.java:1951)
        at com.serenegiant.usb.encoder.biz.H264EncodeConsumer.startMediaCodec(H264EncodeConsumer.java:259)
        at com.serenegiant.usb.encoder.biz.H264EncodeConsumer.run(H264EncodeConsumer.java:136)

宽高都是偶数，正常设置