jiangdongguo/AndroidUSBCamera #54: preview error 打开摄像头，开始预览就报错

xiayouli0122, 2018-06-14:
06-14 17:16:52.015 9549-9580/com.jiangdg.usbcamera V/UVCCameraTextureView: getPreviewTexture:
06-14 17:16:52.016 9549-9575/com.jiangdg.usbcamera I/UVCCameraTextureView: RenderThread#updatePreviewSurface:
06-14 17:16:52.016 9549-9575/com.jiangdg.usbcamera D/UVCCameraTextureView: updatePreviewSurface:release mPreviewSurface
06-14 17:16:52.016 9549-9575/com.jiangdg.usbcamera V/UVCCameraTextureView: updatePreviewSurface:tex_id=1
06-14 17:16:52.018 9549-9570/com.jiangdg.usbcamera V/CameraThread: handleStartPreview:
06-14 17:16:52.018 9549-9570/com.jiangdg.usbcamera I/libUVCCamera: [9570*UVCPreview.cpp:276:callbackPixelFormatChanged]:PIXEL_FORMAT_YUV20SP:
06-14 17:16:52.053 9549-9588/com.jiangdg.usbcamera I/libUVCCamera: [9588*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@YUYV
06-14 17:16:52.068 9549-9590/com.jiangdg.usbcamera I/libUVCCamera: [9590*UVCPreview.cpp:276:callbackPixelFormatChanged]:PIXEL_FORMAT_YUV20SP:
06-14 17:16:53.318 9549-9585/com.jiangdg.usbcamera A/libc: Fatal signal 11 (SIGSEGV), code 2, fault addr 0xdfd0d000 in tid 9585 (CameraThread)

06-14 17:16:53.370 2777-2777/? I/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
    Build fingerprint: 'Android/p201/p201:5.1.1/LMY47V/20171122:userdebug/release-keys'
    Revision: '0'
    ABI: 'arm'
    pid: 9549, tid: 9585, name: CameraThread  >>> com.jiangdg.usbcamera <<<
    signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0xdfd0d000
06-14 17:16:53.383 2777-2777/? I/DEBUG:     r0 dfd0cfcc  r1 ddcce908  r2 0000044c  r3 021c2904
06-14 17:16:53.384 2777-2777/? I/DEBUG:     r4 00000a0c  r5 00000000  r6 ddcce389  r7 00000a0c
        r8 ab06f8b0  r9 ab06fe84  sl ddcce388  fp dfe0aab8
        ip f72a2648  sp dfe0aa70  lr dfe52fcc  pc f725dc7e  cpsr 200f0030
    backtrace:
        #00 pc 00012c7e  /system/lib/libc.so (__memcpy_base+121)
        #01 pc 00014fc8  /data/app/com.jiangdg.usbcamera-1/lib/arm/libuvc.so
06-14 17:16:53.385 2777-2777/? I/DEBUG:     #02 pc 0000c314  /data/app/com.jiangdg.usbcamera-1/lib/arm/libusb100.so (usbi_handle_transfer_completion+320)
        #03 pc 000126a8  /data/app/com.jiangdg.usbcamera-1/lib/arm/libusb100.so
        #04 pc 0000d200  /data/app/com.jiangdg.usbcamera-1/lib/arm/libusb100.so
        #05 pc 0000cb60  /data/app/com.jiangdg.usbcamera-1/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+968)
06-14 17:16:53.386 2777-2777/? I/DEBUG:     #06 pc 0000d3cc  /data/app/com.jiangdg.usbcamera-1/lib/arm/libusb100.so (libusb_handle_events+52)
        #07 pc 00012c20  /data/app/com.jiangdg.usbcamera-1/lib/arm/libuvc.so (_uvc_handle_events+136)
        #08 pc 00016933  /system/lib/libc.so (__pthread_start(void*)+30)
        #09 pc 0001485f  /system/lib/libc.so (__start_thread+6)
06-14 17:16:53.753 2777-2777/? I/DEBUG: Tombstone written to: /data/tombstones/tombstone_02
06-14 17:16:53.755 3106-3126/system_process I/BootReceiver: Copying /data/tombstones/tombstone_02 to DropBox (SYSTEM_TOMBSTONE)
06-14 17:16:53.764 3106-18825/system_process I/UsbCameraManager: /dev/video* num:0
06-14 17:16:53.781 3106-9596/system_process W/ActivityManager:   Force finishing activity 1 com.jiangdg.usbcamera/.view.USBCameraActivity

CPU Info：
Processor : AArch64 Processor rev 4 (aarch64)
processor : 0
Features : fp asimd crc32 wp half thumb fastmult vfp edsp neon vfpv3 tlsi vfpv4 idiva idivt
CPU implementer : 0x41
CPU architecture: 8
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

Hardware : Amlogic

Android OS Version: 5.1.1

求帮忙！

l818b, 2019-01-14:
楼上的兄弟,请问你解决了这个问题没呢?我也遇到了.

l818b, 2019-01-14:
@xiayouli0122 楼上的兄弟,请问你解决了这个问题没呢?我也遇到了.