jiangdongguo/AndroidUSBCamera #62: 两个usb摄像头

gxh-apologize, 2018-07-12:
请问可以同时预览两个usb摄像头吗？或者切换预览也行？

gxh-apologize, 2018-07-12:
怪我没仔细看Demo，两个摄像头切换时可以实现的......请问同时预览怎么实现呢。

amusiaHzr, 2018-09-07:
大兄弟，两路同时预览搞定了么，求指教

nuoyan2, 2018-09-19:
> 怪我没仔细看演示，两个摄像头切换时可以实现的......请问同时预览怎么实现呢。

可以请教下切换怎么实现我只看到requestPermission方法 但是直接调用是不行

nuoyan2, 2018-09-19:
> 怪我没仔细看演示，两个摄像头切换时可以实现的......请问同时预览怎么实现呢。

可以留个联系方式吗

Leessy, 2018-11-28:
我项目一直使用4路同时开的，需要demo可以找我

tang5188, 2018-11-30:
> 我项目一直使用4路同时开的，需要demo可以找我

可以提供个demo给我吗？谢谢大神了！

Leessy, 2018-12-01:
> > 我项目一直使用4路同时开的，需要demo可以找我
> 
> 可以提供个demo给我吗？谢谢大神了！

给你主要逻辑参考
package com.example.mycamarademo;

import android.hardware.usb.UsbDevice;
import android.os.Bundle;
import android.os.Handler;
import android.support.v7.app.AppCompatActivity;
import android.util.Log;
import android.view.TextureView;

import com.serenegiant.usb.Size;
import com.serenegiant.usb.USBMonitor;
import com.serenegiant.usb.UVCCamera;

import java.util.List;

public class MoreActivity111 extends AppCompatActivity {
    private static final String TAG = "MainActivity";
    TextureView mTexture, mTexture2, mTexture3, mTexture4;
    USBMonitor usbMonitor;
    UVCCamera uvcCamera,uvcCamera2,uvcCamera3,uvcCamera4;

    int time;
    Handler handler = new Handler();

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_more);
        mTexture = findViewById(R.id.mTexture);
        mTexture2 = findViewById(R.id.mTexture2);
        mTexture3 = findViewById(R.id.mTexture3);
        mTexture4 = findViewById(R.id.mTexture4);

        mTexture.setRotation(-90);
        mTexture2.setRotation(-90);
        mTexture3.setRotation(-90);
        mTexture4.setRotation(-90);

        usbMonitor = new USBMonitor(this, new USBMonitor.OnDeviceConnectListener() {
            @Override
            public void onAttach(final UsbDevice device) {
                Log.d(TAG, "onAttach: ==" + device.getDeviceName());
                if (device.getDeviceClass() == 239 && device.getDeviceSubclass() == 2) {//根据相机信息选择选需要打开的相机
                    if (device.getDeviceName().contains("usb/001/004") || device.getDeviceName().contains("usb/003/004")
                            || device.getDeviceName().contains("usb/001/003") || device.getDeviceName().contains("usb/003/003")) {
                        
                        //此处请求权限，需点击确定，有系统权限可忽略
                        //每个注册权限增加延时，如果4个摄像同时注册权限 可能权限弹窗只会显示一个，导致其他的相机权限未确定，
                        handler.postDelayed(new Runnable() {
                            @Override
                            public void run() {
                                Log.d(TAG, "run: 发送 dev=" + device.getDeviceName());
                                usbMonitor.requestPermission(device);
                            }
                        }, time++ * 500);
                    }

                }
            }

            @Override
            public void onDettach(UsbDevice usbDevice) {

            }

            @Override
            public void onConnect(UsbDevice usbDevice, USBMonitor.UsbControlBlock ctrlBlock, boolean b) {
                new Thread(new myRunnable(ctrlBlock)).start();
            }

            @Override
            public void onDisconnect(UsbDevice usbDevice, USBMonitor.UsbControlBlock usbControlBlock) {
                // TODO: 2018/12/1 根据pid vid 或name判断哪个相机断开连接 
            }

            @Override
            public void onCancel(UsbDevice usbDevice) {

            }
        });
        usbMonitor.register();
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        try {
            usbMonitor.unregister();
            uvcCamera.stopPreview();
            uvcCamera.close();
//            uvcCamera2.close();
//            uvcCamera3.close();
        } catch (Exception e) {

        }

    }

    class myRunnable implements Runnable {
        USBMonitor.UsbControlBlock usbControlBlock;

        public myRunnable(USBMonitor.UsbControlBlock usbControlBlock) {
            this.usbControlBlock = usbControlBlock;
        }

        @Override
        public void run() {
            final UVCCamera camera = new UVCCamera();
            try {
                camera.open(usbControlBlock);
            } catch (Exception e) {
                Log.d(TAG, "开启相机错误！！！！" + camera.getDeviceName());
                return;
            }

            List<Size> supportedSizeList = camera.getSupportedSizeList();
            if (supportedSizeList != null) {
                for (Size size : supportedSizeList) {
                    Log.d(TAG, "run: size=" + size.width + "---" + size.height);
                }
            }

            setCameraParameter(camera);
            //根据不同相机接入name  或者根据pid vid 指定相机在那个view显示
            if (usbControlBlock.getDeviceName().contains("usb/001/004")) {
                camera.setPreviewTexture(mTexture.getSurfaceTexture());
            } else if (usbControlBlock.getDeviceName().contains("usb/001/003")) {
                camera.setPreviewTexture(mTexture2.getSurfaceTexture());
            } else if (usbControlBlock.getDeviceName().contains("usb/003/004")) {
                camera.setPreviewTexture(mTexture3.getSurfaceTexture());
            } else if (usbControlBlock.getDeviceName().contains("usb/003/003")) {
                camera.setPreviewTexture(mTexture4.getSurfaceTexture());
//                camera.setFrameCallback(iFrameCallback, UVCCamera.PIXEL_FORMAT_YUV420SP);//设置数据回调
                uvcCamera = camera;
            }
            camera.startPreview();
        }
    }

    //相机参数  可根据不同相机设置不同参数，如果一个OTG接入多个相机 需注意带宽和帧率设置，否则出现带宽不够导致相机不显示数据
    private void setCameraParameter(UVCCamera camera) {
        try {
            //设置预览尺寸 根据设备自行设置
            camera.setPreviewSize(640,
                    480,
                    1,
                    15,
                    UVCCamera.FRAME_FORMAT_YUYV,
//                    UVCCamera.FRAME_FORMAT_MJPEG,//此格式设置15帧生效
                    0.4f);
            Log.d(TAG, "**设置参数成功=" + camera.getDeviceName());
        } catch (final IllegalArgumentException e) {
            Log.d(TAG, "**设置参数失败=" + camera.getDeviceName());
            return;
        }
    }
}

gxh-apologize, 2018-12-03:
额，我们最终还是单路预览的，，这个项目挺久了，忘记了close issue，感谢楼上提供的思路及代码。

closed 2018-12-03 by gxh-apologize

tang5188, 2018-12-03:
代码能行得通，但是在我这边始终都不行。使用这个方式，我可以单独预览A或B摄像头，也可以切换预览，但总不能同时预览。我这边是单路USB2.0，有可能是硬件上的问题了。谢谢 Leessy 的方案思路！

Leessy, 2018-12-03:
> 代码能行得通，但是在我这边始终都不行。使用这个方式，我可以单独预览A或B摄像头，也可以切换预览，但总不能同时预览。我这边是单路USB2.0，有可能是硬件上的问题了。谢谢 Leessy 的方案思路！

注意 camera.setPreviewSize方法，如果两个相机是通过一个OTG接入的，需要注意带宽和预览格式等，而且有的摄像头不支持YUV格式，根据你说的分别能打开，很有可能就是带宽问题

tang5188, 2018-12-03:
> > 代码能行得通，但是在我这边始终都不行。使用这个方式，我可以单独预览A或B摄像头，也可以切换预览，但总不能同时预览。我这边是单路USB2.0，有可能是硬件上的问题了。谢谢 Leessy 的方案思路！
> 
> 注意 camera.setPreviewSize方法，如果两个相机是通过一个OTG接入的，需要注意带宽和预览格式等，而且有的摄像头不支持YUV格式，根据你说的分别能打开，很有可能就是带宽问题

嗯，确实可能是带宽问题。我调整了分辨率跟带宽的参数，也是行不通的，可能是USB2.0的带宽有点少，带不动2个同时工作的摄像头。

yanming0909, 2019-01-07:
> 给你主要逻辑参考
> package com.example.mycamarademo;
>
> import android.hardware.usb.UsbDevice;
> import android.os.Bundle;
> import android.os.Handler;
> import android.support.v7.app.AppCompatActivity;
> import [...]

请问双摄像录制音频问题，录音之后只有一路摄像头有录制视频，另一路是黑屏的  大佬知道怎么回事吗

Leessy, 2019-01-07:
> 请问双摄像录制音频问题，录音之后只有一路摄像头有录制视频，另一路是黑屏的 大佬知道怎么回事吗

我没做过这样的需求，但是你说录音之后另一路是黑屏，意思是录音之前是两路都是视频的？ 可以看下相机回调数据的地方代码是否有问题，另附uvc库原著demo  https://github.com/saki4510t/UVCCamera

yanming0909, 2019-02-17:
> 怪我没仔细看Demo，两个摄像头切换时可以实现的......请问同时预览怎么实现呢。

请问下切换摄像头实现怎么操作，大佬

yanming0909, 2019-02-17:
> 我没做过这样的需求，但是你说录音之后另一路是黑屏，意思是录音之前是两路都是视频的？ 可以看下相机回调数据的地方代码是否有问题，另附uvc库原著demo https://github.com/saki4510t/UVCCamera

好像是MIC只能打开一个  打开2次就出问题  现在这需求不要了  改为切换摄像头 大佬有实现方案吗
new2个camera 我这边出问题了，预览不了了

Leessy, 2019-02-17:
> > 怪我没仔细看Demo，两个摄像头切换时可以实现的......请问同时预览怎么实现呢。
> 
> 请问下切换摄像头实现怎么操作，大佬

首先你要明白摄像头是可以多个同时启动的 且互不影响，可以创建多个uvccamera,uvccamera也是可以关闭的，需求可以理解为同时启动两个摄像头并切换显示，也可以理解为启动一个摄像头，在切换时关闭当前的摄像头启动另一个摄像头，两种方式都差不多，同时预览两个或以上相机只需要把每个uvccamera.setPreviewTexture(mTexture.getSurfaceTexture());这个方法分别设置需要显示的textureview就可以了， 切换有预览只需要重新调用setPreviewTexture设置成其他的显示view即可，相机也可以在后台开启只需要回调数据在需要的时候才预览，setPreviewTexture这个只需要new一个SurfaceTexture传入即可，在需要显示的时候在设置view的SurfaceTexture

yanming0909, 2019-02-17:
> 首先你要明白摄像头是可以多个同时启动的 [...]

切换实现了已经可以 现在就是偶尔第一次打开无预览，切换2下就有了，再仔细看看 感谢

mafanwei, 2020-01-12:
> 切换实现了已经可以 现在就是偶尔第一次打开无预览，切换2下就有了，再仔细看看 感谢

可以提供个切换demo给我么，感谢！

Leessy, 2020-01-12:
请问需要什么样的demo ？关于usb相机我项目仓库有一个封装，在coolkotkin 下，有一些
简单的说明

mafanwei, 2020-01-12:
> 请问需要什么样的demo ？关于usb相机我项目仓库有一个封装，在coolkotkin 下，有一些 简单的说明
> […](#)
> Sent from my iPhone

我想要一个可以切换usb相机的demo，您有么？

Leessy, 2020-01-12:
不明白需求，相机本身就是单独的设备，你可以开多个多窗口分别显示，也可以开一个相机多窗口显示，切换usb相机 又是什么意思呢？多相机共用一个窗口显示，我封装的库可以很简单的实现以上需求

发自我的iPhone

mafanwei, 2020-01-12:
> 不明白需求，相机本身就是单独的设备，你可以开多个多窗口分别显示，也可以开一个相机多窗口显示，切换usb相机 又是什么意思呢？多相机共用一个窗口显示，我封装的库可以很简单的实现以上需求 发自我的iPhone
> […](#)

项目的需求是这样的：连接一个usb hub，上面有3个摄像头，可以随时切换上面的摄像头显示，始终只有一个摄像头在工作，我不需要预览，我只需要其中的相机返回的数据就可以了。
封装库是在您的coolkotlin下么？我似乎没有看到有关它的代码。

Leessy, 2020-01-12:
照你意思不就是多个相机，想开哪个开哪个，不需要同时开就关掉再开另一个就行了

mafanwei, 2020-01-12:
> 照你意思不就是多个相机，想开哪个开哪个，不需要同时开就关掉再开另一个就行了
> […](#)
> Sent from my iPhone

是的。我好像在您的coolkotlin中找到它了，我现在尝试一下它是否可用。

mafanwei, 2020-01-12:
我尝试了您的demo，但是仍无法全部切换，只能预览第一个/第二个，切换就会报错：[14706*stream.c:1605:uvc_stream_start_bandwidth]:fail
看起来是因为带宽不足，我调用了it?.destroyCamera()，情况仍无好转，我应该怎么办？

mafanwei, 2020-01-12:
> 照你意思不就是多个相机，想开哪个开哪个，不需要同时开就关掉再开另一个就行了
> […](#)
> Sent from my iPhone

我尝试了您的demo，但是仍无法全部切换，只能预览第一个/第二个，切换就会报错：[14706*stream.c:1605:uvc_stream_start_bandwidth]:fail
看起来是因为带宽不足，我调用了it?.destroyCamera()，情况仍无好转，我应该怎么办？
我的代码如下：
```
 fun changeUseCameraDevice() {
       // closeCamera()
        nowUSB++
        if(nowUSB>=CamerasMng.cameraList.size) {
            nowUSB = 0
        }
        val name = CamerasMng.cameraList[nowUSB].devName
        CamerasMng.cameraList.forEach {
            Log.d("CamerasMng", "CamerasMng pid ${it.pid}" +"device name"+it.devName)
            if (it.devName == name) {//选择指定相机进行启动
                var c = it
                val b = c?.openCamera()
                c?.setFrameCall(object : IFrameCall{override fun call(bf: ByteBuffer, w: Int, h: Int) {
                    mVideoCapture.onUsbCameraAvailable(decodeValue(bf), w, h)
                }})//设置相机数据帧回调
                // c?.startPreview(TextureView(mContext))//启动相机预览
                c?.startPreview()//启动相机--不需预览只需要回调数据的方式

            } else {
                it?.destroyCamera()
            }
        }
    }
```

Leessy, 2020-01-12:
异常信息为带宽不足，明显是同时开了多个相机，并且导致后面开起的带宽异常，你应该设置更小的分辨率或者帧率，也可以设置带宽占用更低的格式，比如jpeg 等，还可以指定单个相机的带宽占用比。这个你们的andriod设备和相机有关

发自我的iPhone

 

  
  fun changeUseCameraDevice() {        // closeCamera()         nowUSB++         if(nowUSB&gt;=CamerasMng.cameraList.size) {             nowUSB = 0         }         val name = CamerasMng.cameraList[nowUSB].devName         CamerasMng.cameraList.forEach {             Log.d("CamerasMng", "CamerasMng pid ${it.pid}" +"device name"+it.devName)             if (it.devName == name) {//选择指定相机进行启动                 var c = it                 val b = c?.openCamera()                 c?.setFrameCall(object : IFrameCall{override fun call(bf: ByteBuffer, w: Int, h: Int) {                     mVideoCapture.onUsbCameraAvailable(decodeValue(bf), w, h)                 }})//设置相机数据帧回调                 // c?.startPreview(TextureView(mContext))//启动相机预览                 c?.startPreview()//启动相机--不需预览只需要回调数据的方式             } else {                 it?.destroyCamera()             }         }     }

mafanwei, 2020-01-12:
> 异常信息为带宽不足，明显是同时开了多个相机，并且导致后面开起的带宽异常，你应该设置更小的分辨率或者帧率，也可以设置带宽占用更低的格式，比如jpeg 等，还可以指定单个相机的带宽占用比。这个你们的andriod设备和相机有关 发自我的iPhone
> […](#)

指定单个相机的带宽占用比，您说的很对，我指定了每个camera利用30%的带宽，问题就成功解决了。谢谢您！

Leessy, 2020-01-12:
如果确实有很多相关需求，建议直接学习原作者的来源项目，我这里也只是简单应用

发自我的iPhone

mafanwei, 2020-01-12:
> 如果确实有很多相关需求，建议直接学习原作者的来源项目，我这里也只是简单应用 发自我的iPhone
> […](#)

是的，可是我并不是一个硬件/驱动/framework/系统工程师，只是单纯的碰到了项目需要用到这个，我有尝试过其他方案，也自己试着封装了一下最新的uvccamera，但是公司的项目用的是cmake，camke是没法和ndk-build并存的，而我对c层又很苦手，所以没办法，只能寻求他人封装好的库二次利用。您封装的并不是最新的库，缺少64位的so包，不过现阶段来看，已经是最好的解决办法了，或许以后有机会我会重新弄一弄这块。
再次谢谢您！

Leessy, 2020-01-12:
原作者的项目可以直接编译，并附带各种使用demo ，根本不需要熟悉cpp，只需要参考学习JAVA层的应用即可。我有编译64位的so，只是我没有使用

发自我的iPhone

mafanwei, 2020-01-12:
> 原作者的项目可以直接编译，并附带各种使用demo ，根本不需要熟悉cpp，只需要参考学习JAVA层的应用即可。我有编译64位的so，只是我没有使用 发自我的iPhone
> […](#)

我加您好友吧，原作者的项目似乎因为我的ndk版本问题，我未成功编译

Jack-Hu, 2021-06-25:
> 给你主要逻辑参考
> package com.example.mycamarademo;
>
> import android.hardware.usb.UsbDevice;
> import android.os.Bundle;
> import android.os.Handler;
> import android.support.v7.app.AppCompatActivity;
> import [...]

请教个问题，用您的这个代码，无法多路预览，只能单路。关键log如下
W/libUVCCamera: [4737*UVCPreview.cpp:119:get_frame]:allocate new frame
W/libUVCCamera: [4736*UVCPreview.cpp:119:get_frame]:allocate new frame
E/BufferQueueProducer: [SurfaceTexture-1-4699-2] connect: BufferQueue has been abandoned
E/BufferQueueProducer: [SurfaceTexture-1-4699-2] connect: BufferQueue has been abandoned
E/BufferQueueProducer: [SurfaceTexture-1-4699-2] connect: BufferQueue has been abandoned
E/libuvc/stream: [4741*stream.c:1605:uvc_stream_start_bandwidth]:fail
W/libUVCCamera: [4741*diag.c:88:uvc_perror]:failed start_streaming:I/O error (-1)
E/BufferQueueProducer: [SurfaceTexture-1-4699-2] connect: BufferQueue has been abandoned
E/BufferQueueProducer: [SurfaceTexture-1-4699-2] connect: BufferQueue has been abandoned
E/BufferQueueProducer: [SurfaceTexture-1-4699-2] connect: BufferQueue has been abandoned
...

全都是 connect: BufferQueue has been abandoned 这个，看了几篇说把SurfaceTexture提示为成员也没用。不知道啥问题
可以给个完整demo的地址吗

Leessy, 2021-06-25:
github/coolkotlin 我这个项目下有关于uvc 相机的封装，可以参考一下，另外多个相机都共用usb口的话注意带宽问题

chenYuCui, 2022-05-10:
> 
我想个demo，大神，怎么联系

Jack-Hu, 2022-05-11:
这是虹软官方提供的

Leessy, 2022-10-11:
参考这个库，项目里也有实现，代码比较乱，平时写的demohttps://github.com/Leessy/CoolKotlin/blob/master/README-UsbCamera.md

gatspy, 2024-08-27:
mark下，需要指定摄像头