jiangdongguo/AndroidUSBCamera #66: 大神您好,请问我连接公司摄像头后,无法从IFrameCallback获取到frame(YUV) 数据,但是连接其他摄像头可以,请问是什么原因呢?

1159658610, 2018-07-28:


SunPointed, 2018-08-23:
你们公司的摄像头是不是不止包含一个usb device，这个demo中的UVCCameraHelper默认使用第一个设备
`public void requestPermission(int index) {
        List<UsbDevice> devList = getUsbDeviceList();
        if (devList == null || devList.size() == 0) {
            return;
        }
        int count = devList.size();
        if (index >= count)
            new IllegalArgumentException("index illegal,should be < devList.size()");
        if (mUSBMonitor != null) {
            mUSBMonitor.requestPermission(getUsbDeviceList().get(index));
        }
    }`
改成这样
`public void requestPermission(UsbDevice device) {
        if(device == null){
            return;
        }
        if (mUSBMonitor != null) {
            mUSBMonitor.requestPermission(device);
        }
    }`
然后在USBCameraActivity中，你需要根据设备的product Id去判断是不是usb摄像头，是才requestPermission，如下：
`@Override
        public void onAttachDev(UsbDevice device) {
            if (mCameraHelper == null || mCameraHelper.getUsbDeviceCount() == 0) {
                showShortMsg("check no usb camera");
                return;
            }

            if (device.getProductId() != '你需要usb设备的product id') {
                return;
            }
            // request open permission
            if (!isRequest) {
                isRequest = true;
                if (mCameraHelper != null) {
                    mCameraHelper.requestPermission(device);
                }
            }
        }`

1159658610, 2018-09-03:
非常感谢您详细的解答,我可以试试你这种方法,我目前是这样解决,因为我们只需要获取到数据就可以了.
public void handleStartPreview(final Object surface) {
            if (DEBUG) Log.e(TAG_THREAD, "handleStartPreview:");
            //if ((mUVCCamera == null) || mIsPreviewing) return;
            try {
                mUVCCamera.setPreviewSize(mWidth, mHeight, 1, 31, UVCCamera.DEFAULT_PREVIEW_MODE, mBandwidthFactor);
                mUVCCamera.setFrameCallback(mIFrameCallback, UVCCamera.PIXEL_FORMAT_RAW);

            } catch (final IllegalArgumentException e) {
                try {
...
}
这是修改的主要部分.

WestbrookZero, 2019-05-14:
你好，麻烦问一下，如何从IFrameCallback的onFrame函数中获取YUV 中的Y值数据？

SunPointed, 2019-05-21:
研究下YUV格式在数组中如何存储。。。