jiangdongguo/AndroidUSBCamera #72: 录像时程序崩溃退出

MaYunan, 2018-08-27:
当我成功打开预览，然后点击录像时报程序异常退出。检查错误是一个native_dequeueOutputBuffer()方法出错，请问是怎么回事呢

jiangdongguo, 2018-08-28:
换台手机试试，怀疑手机不支持指定的编码器颜色格式

MaYunan, 2018-09-06:
我下载了这个项目https://github.com/saki4510t/UVCCamera，里面的程序就可以完成录制视频。还想请教一下如果我的USB摄像头上有MIC，怎么才能使用这个MIC实现录音呢

mzc007, 2019-09-04:
一录制视频就闪退，报这两个错
E/H264EncodeConsumer: couldn't find a good color format for OMX.MTK.VIDEO.ENCODER.AVC / video/avc
E/H264EncodeConsumer: Unable to find an appropriate codec for video/avc

jiangdongguo, 2019-09-04:
> 一录制视频就闪退，报这两个错
> E/H264EncodeConsumer: couldn't find a good color format for OMX.MTK.VIDEO.ENCODER.AVC / video/avc
> E/H264EncodeConsumer: Unable to find an appropriate codec for video/avc

你的设备不支持编码器 video/avc

heaveninfosoft, 2020-10-28:
@jiangdongguo video is blue/violet color
this because of video encoder can you change it??
problem in video encoder please change

mzc007, 2021-01-23:
请问总是报这个错是什么原因，调用的时候已经判空了  为什么还会这样