jiangdongguo/AndroidUSBCamera #74: nativeStopPreview(mNativePtr)报错关闭异常

ztc464728422, 2018-08-30:
场景是：快速关闭释放摄像头后重新创建，关闭预览处jni方法报错
nativeStopPreview(mNativePtr);

08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410] JNI DETECTED ERROR IN APPLICATION: jmethodID was NULL
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]     in call to CallVoidMethodV
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410] "Thread-4583" prio=5 tid=30 Runnable
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   | group="main" sCount=0 dsCount=0 obj=0x13546160 self=0x7f57aa1a00
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   | sysTid=19600 nice=0 cgrp=default sched=0/0 handle=0x7f5d807440
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   | state=R schedstat=( 12813919206 190641066 5107 ) utm=1252 stm=29 core=4 HZ=100
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   | stack=0x7f5d70b000-0x7f5d70d000 stackSize=1013KB
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   | held mutexes= "mutator lock"(shared held)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #00 pc 000000000048bce8  /system/lib64/libart.so (_ZN3art15DumpNativeStackERNSt3__113basic_ostreamIcNS0_11char_traitsIcEEEEiPKcPNS_9ArtMethodEPv+236)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #01 pc 000000000045ac54  /system/lib64/libart.so (_ZNK3art6Thread4DumpERNSt3__113basic_ostreamIcNS1_11char_traitsIcEEEE+220)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #02 pc 000000000030ed50  /system/lib64/libart.so (_ZN3art9JavaVMExt8JniAbortEPKcS2_+1000)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #03 pc 000000000030f608  /system/lib64/libart.so (_ZN3art9JavaVMExt9JniAbortVEPKcS2_St9__va_list+116)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #04 pc 000000000014217c  /system/lib64/libart.so (_ZN3art11ScopedCheck6AbortFEPKcz+144)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #05 pc 000000000014a728  /system/lib64/libart.so (_ZN3art11ScopedCheck5CheckERNS_18ScopedObjectAccessEbPKcPNS_12JniValueTypeE.constprop.116+11972)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #06 pc 000000000015bc30  /system/lib64/libart.so (_ZN3art8CheckJNI11CallMethodVEPKcP7_JNIEnvP8_jobjectP7_jclassP10_jmethodIDSt9__va_listNS_9Primitive4TypeENS_10InvokeTypeE+684)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #07 pc 000000000015def8  /system/lib64/libart.so (_ZN3art8CheckJNI15CallVoidMethodVEP7_JNIEnvP8_jobjectP10_jmethodIDSt9__va_list+68)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #08 pc 0000000000013e08  /data/app/synway.osc-2/lib/arm64/libUVCCamera.so (_ZN7_JNIEnv14CallVoidMethodEP8_jobjectP10_jmethodIDz+120)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #09 pc 0000000000013c58  /data/app/synway.osc-2/lib/arm64/libUVCCamera.so (_ZN10UVCPreview19do_capture_callbackEP7_JNIEnvP9uvc_frame+188)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #10 pc 0000000000013930  /data/app/synway.osc-2/lib/arm64/libUVCCamera.so (_ZN10UVCPreview10do_captureEP7_JNIEnv+108)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #11 pc 0000000000013330  /data/app/synway.osc-2/lib/arm64/libUVCCamera.so (_ZN10UVCPreview19capture_thread_funcEPv+60)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #12 pc 0000000000065fc4  /system/lib64/libc.so (_ZL15__pthread_startPv+52)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   native: #13 pc 000000000001edc4  /system/lib64/libc.so (__start_thread+16)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410]   (no managed stack frames)
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/java_vm_ext.cc:410] 
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/runtime.cc:368] Runtime aborting...
08-30 09:08:30.294 19367-19600/? A/art: art/runtime/runtime.cc:368] 
08-30 09:08:30.294 19367-19600/? A/libc: Fatal signal 6 (SIGABRT), code -6 in tid 19600 (Thread-4583)
08-30 09:08:30.354 579-579/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
08-30 09:08:30.354 579-579/? A/DEBUG: Build fingerprint: 'samsung/c9ltezc/c9ltechn:6.0.1/MMB29M/C9000ZCU1AQG1:user/release-keys'
08-30 09:08:30.354 579-579/? A/DEBUG: Revision: '9'
08-30 09:08:30.354 579-579/? A/DEBUG: ABI: 'arm64'
08-30 09:08:30.354 579-579/? A/DEBUG: pid: 19367, tid: 19600, name: Thread-4583  >>> synway.osc <<<
08-30 09:08:30.354 579-579/? A/DEBUG: signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
08-30 09:08:30.384 579-579/? A/DEBUG: Abort message: 'art/runtime/java_vm_ext.cc:410] JNI DETECTED ERROR IN APPLICATION: jmethodID was NULL'
08-30 09:08:30.384 579-579/? A/DEBUG:     x0   0000000000000000  x1   0000000000004c90  x2   0000000000000006  x3   0000000000000000
08-30 09:08:30.384 579-579/? A/DEBUG:     x4   0000000000000000  x5   0000000000000001  x6   0000000000000000  x7   0000000000000000
08-30 09:08:30.384 579-579/? A/DEBUG:     x8   0000000000000083  x9   0000000000007500  x10  0000000001c43dc8  x11  0000000001c43f08
08-30 09:08:30.384 579-579/? A/DEBUG:     x12  0000000000000000  x13  0000007f7caef000  x14  0000000000000000  x15  0000007f7cae1000
08-30 09:08:30.384 579-579/? A/DEBUG:     x16  0000007f7cae1568  x17  0000007f7ca733dc  x18  0000000000000000  x19  0000007f5d807500
08-30 09:08:30.384 579-579/? A/DEBUG:     x20  0000007f5d807440  x21  000000000000006e  x22  0000000000000006  x23  0000007f579c7500
08-30 09:08:30.384 579-579/? A/DEBUG:     x24  0000007f78dad680  x25  0000007f57aa1a00  x26  0000007f790e1f00  x27  0000000000000001
08-30 09:08:30.384 579-579/? A/DEBUG:     x28  0000007f579c73c0  x29  0000007f5d806940  x30  0000007f7ca70b78
08-30 09:08:30.384 579-579/? A/DEBUG:     sp   0000007f5d806940  pc   0000007f7ca733e4  pstate 0000000020000000
08-30 09:08:30.394 579-579/? A/DEBUG: backtrace:
08-30 09:08:30.394 579-579/? A/DEBUG:     #00 pc 00000000000693e4  /system/lib64/libc.so (tgkill+8)
08-30 09:08:30.394 579-579/? A/DEBUG:     #01 pc 0000000000066b74  /system/lib64/libc.so (pthread_kill+68)
08-30 09:08:30.394 579-579/? A/DEBUG:     #02 pc 00000000000238e8  /system/lib64/libc.so (raise+28)
08-30 09:08:30.394 579-579/? A/DEBUG:     #03 pc 000000000001e218  /system/lib64/libc.so (abort+60)
08-30 09:08:30.394 579-579/? A/DEBUG:     #04 pc 0000000000431dd8  /system/lib64/libart.so (_ZN3art7Runtime5AbortEv+304)
08-30 09:08:30.394 579-579/? A/DEBUG:     #05 pc 0000000000136364  /system/lib64/libart.so (_ZN3art10LogMessageD2Ev+3136)
08-30 09:08:30.394 579-579/? A/DEBUG:     #06 pc 000000000030f188  /system/lib64/libart.so (_ZN3art9JavaVMExt8JniAbortEPKcS2_+2080)
08-30 09:08:30.394 579-579/? A/DEBUG:     #07 pc 000000000030f608  /system/lib64/libart.so (_ZN3art9JavaVMExt9JniAbortVEPKcS2_St9__va_list+116)
08-30 09:08:30.394 579-579/? A/DEBUG:     #08 pc 000000000014217c  /system/lib64/libart.so (_ZN3art11ScopedCheck6AbortFEPKcz+144)
08-30 09:08:30.394 579-579/? A/DEBUG:     #09 pc 000000000014a728  /system/lib64/libart.so (_ZN3art11ScopedCheck5CheckERNS_18ScopedObjectAccessEbPKcPNS_12JniValueTypeE.constprop.116+11972)
08-30 09:08:30.394 579-579/? A/DEBUG:     #10 pc 000000000015bc30  /system/lib64/libart.so (_ZN3art8CheckJNI11CallMethodVEPKcP7_JNIEnvP8_jobjectP7_jclassP10_jmethodIDSt9__va_listNS_9Primitive4TypeENS_10InvokeTypeE+684)
08-30 09:08:30.394 579-579/? A/DEBUG:     #11 pc 000000000015def8  /system/lib64/libart.so (_ZN3art8CheckJNI15CallVoidMethodVEP7_JNIEnvP8_jobjectP10_jmethodIDSt9__va_list+68)
08-30 09:08:30.394 579-579/? A/DEBUG:     #12 pc 0000000000013e08  /data/app/synway.osc-2/lib/arm64/libUVCCamera.so (_ZN7_JNIEnv14CallVoidMethodEP8_jobjectP10_jmethodIDz+120)
08-30 09:08:30.394 579-579/? A/DEBUG:     #13 pc 0000000000013c58  /data/app/synway.osc-2/lib/arm64/libUVCCamera.so (_ZN10UVCPreview19do_capture_callbackEP7_JNIEnvP9uvc_frame+188)
08-30 09:08:30.394 579-579/? A/DEBUG:     #14 pc 0000000000013930  /data/app/synway.osc-2/lib/arm64/libUVCCamera.so (_ZN10UVCPreview10do_captureEP7_JNIEnv+108)
[... cut ...]
08-30 09:08:45.404 3352-3355/? E/ANDR-PERF-MPCTL: Invalid profile no. 0, total profiles 0 only
08-30 09:08:47.064 1161-1977/? E/Watchdog: !@Sync 1968 [08-30 09:08:47.066]
08-30 09:09:00.014 438-438/? E/lowmemorykiller: Error writing /proc/13985/oom_score_adj; errno=22
08-30 09:09:01.104 19752-19752/? E/Zygote: v2
08-30 09:09:01.104 19752-19752/? E/Zygote: accessInfo : 0
08-30 09:09:08.254 14784-14784/? E/dhcpcd: wlan0: rebind time greater than lease time, forcing to 31500 seconds
08-30 09:09:08.254 655-772/? E/QC-NETMGR-LIB: unrecognized ifindex 17

closed 2018-08-31 by ztc464728422

zhuwei, 2019-05-13:
请问你是怎么解决的？我这边快速关闭然后创建，没报错，但也没图像

hushengjun, 2019-10-09:
这个问题关闭了，是已经解决了吗？我还是出现了同样的问题

zhuwei, 2019-10-09:
@hushengjun 只能想其他办法解决，加个过度activity

androidscx, 2022-02-10:
怎么解决这个问题