jiangdongguo/AndroidUSBCamera #77: 有没有办法接入多路usb摄像头

amusiaHzr, 2018-09-06:
如题，项目中想要接入双路usb摄像头，有大神提供解题思路么

lingxiaopua, 2018-09-20:
初始化用的是单例模式，有几个摄像头就new几个，就这么简单

amusiaHzr, 2018-10-23:
> 初始化用的是单例模式，有几个摄像头就new几个，就这么简单

是的，修改一下单例就行了，谢谢！

Leessy, 2018-11-28:
一直用的作者原版，项目用的是同时开4路相机

vincent8877, 2018-12-06:
有人双路实现了嘛？求指导具体实现细节啊，小白不太懂。跪求！在线等

vincent8877, 2018-12-06:
@amusiaHzr @lingxiaopua 两位大神求指导

vincent8877, 2018-12-06:
>           一直用的作者原版，项目用的是同时开4路相机
你说的原版是saki4510t的嘛？

vincent8877, 2018-12-06:
>           一直用的作者原版，项目用的是同时开4路相机

@Leessy 你说的原版是saki4510t的嘛？

Leessy, 2018-12-06:
> > ```
> >       一直用的作者原版，项目用的是同时开4路相机
> > ```
> 
> @Leessy 你说的原版是saki4510t的嘛？
是的

Leessy, 2018-12-06:
> > ```
> >       一直用的作者原版，项目用的是同时开4路相机
> > ```
> 
> @Leessy 你说的原版是saki4510t的嘛？
我在这个 issues下有回答一个方案
https://github.com/jiangdongguo/AndroidUSBCamera/issues/62#issuecomment-443392620

frank0123, 2019-01-03:
@Leessy 您好！我最近也在使用这个项目进行“多路USB摄像头”开发，使用的是作者最新的源码！我也仔细看过了您在 #62 中的回答方案，但是这个方案好像是作者以前旧版的代码；我也清楚虽然内容有些不同，但是逻辑是一样的！个人使用现在的代码在进行“多路”开发时碰到的问题是，我不太明白现有的UVCCameraHelper和您在旧版中使用的UsbMonitor具体的区别在哪里，有一些细节也不太一样，无法参照您提供的旧有的逻辑代码进行编写！

不太清楚您对现在最新的逻辑代码是否了解，不知道您能对此提供一些建议给我吗？十分感谢！

PS：我需要实现的是3路USB摄像头接入，虽然可以不需要预览，但是我需要每一路的图像数据分别进行处理！

Leessy, 2019-01-03:
> @Leessy 您好！我最近也在使用这个项目进行“多路USB摄像头”开发，使用的是作者最新的源码！我也仔细看过了您在 #62 中的回答方案，但是这个方案好像是作者以前旧版的代码；我也清楚虽然内容有些不同，但是逻辑是一样的！个人使用现在的代码在进行“多路”开发时碰到的问题是，我不太明白现有的UVCCameraHelper和您在旧版中使用的UsbMonitor具体的区别在哪里，有一些细节也不太一样，无法参照您提供的旧有的逻辑代码进行编写！
> 
> 不太清楚您对现在最新的逻辑代码是否了解，不知道您能对此提供一些建议给我吗？十分感谢！
> 
> PS：我需要实现的是3路USB摄像头接入(3个同一型号的USB摄像头通过一个USB-HUB连接手机)，虽然可以不需要预览，但是我需要每一路的图像数据分别进行处理！

uvc源码中是没有UVCCameraHelper这个类的，我实现方式参考demo在这里https://github.com/saki4510t/UVCCamera  ，UsbMonitor 主要控制需要启动的设备，分别opencamera后设置显示的surface和callback就可了，实现方式还是#62回答那个，注意事项也有说明，带宽和参数设置等

frank0123, 2019-01-03:
@Leessy 抱歉，我理解错“原意”！其实我想说的是，能否根据作者现在最新的逻辑代码添加多路USB接入功能呢？

PS：我现在正在尝试使用您提供的逻辑代码进行编码，有一个问题想请教您一下，就是在class myRunnable 中这个赋值uvcCamera = camera;是硬编码的吗？如果不是，那么剩下的 uvcCamera2/3/4 怎么进行动态赋值呢？我不是很理解这一块的逻辑，还希望您能否指导一下，十分感谢！

cyixlq, 2020-09-01:
> > 初始化用的是单例模式，有几个摄像头就new几个，就这么简单
> 
> 是的，修改一下单例就行了，谢谢！

您好，能详细说一下你是怎么改的吗？我这边也需要同时开好几个摄像头