jiangdongguo/AndroidUSBCamera #81: 太奇怪了...总是找不到类

ylmyg, 2018-10-19:
错误: 无法访问IAspectRatioView
找不到com.serenegiant.widget.IAspectRatioView的类文件
错误: 找不到符号
符号:   变量 super
位置: 类 USBCameraActivity
代码里面是有的呀，为什么运行的时候总是找不到？

jiangdongguo, 2018-10-22:
工程gradle加了这个没？
repositories {
        maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
        jcenter()
    }

ylmyg, 2018-10-23:
已经加了呀，就是下载你的代码来运行的，代码我也没改变什么，我是用AS 3.1.3 Gradle 4.4
就是报错：
错误: 无法访问IAspectRatioView 找不到com.serenegiant.widget.IAspectRatioView的类文件 |  
-- | --
错误: 找不到符号 符号:   变量 super 位置: 类 USBCameraActivity |  
错误: 找不到符号 符号:   变量 this 位置: 类 USBCameraActivity |  
错误: 找不到符号 符号:   变量 this 位置: 类 USBCameraActivity |  
错误: 找不到符号 符号:   变量 super 位置: 类 USBCameraActivity |  
错误: 找不到符号 符号:   变量 super 位置: 类 USBCameraActivity |  
错误: 找不到符号 符号:   变量 super 位置: 类 USBCameraActivity |  
错误: 不是封闭类: USBCameraActivity |  
错误: 不是封闭类: USBCameraActivity |  
错误: 找不到符号 符号:   变量 this 位置: 类 USBCameraActivity |  
错误: 方法不会覆盖或实现超类型的方法 |  
错误: 方法不会覆盖或实现超类型的方法 |

1159658610, 2018-10-24:
这个好像是封装在jar包中的,我之前也遇到过,不过我是根据AS最上面的提示找到对应jar包,然后添加到项目中就行了.

1159658610, 2018-10-24:
C:\Users\Administrator\.gradle\caches\transforms-1\files-1.1\common-1.5.20.aar\68e488dd628b30ff00362fd19ef28d6d\jars
这是我电脑对应的jar包位置,如果还没找到错误原因,你可以尝试下.

hnlht, 2020-05-28:
> 重新再下一次项目，编译又没有错误了。。
你好，你知道怎么解决的吗？我是直接下载的aar依赖到项目里的。也报这个找不到com.serenegiant.widget.IAspectRatioView的类文件错误，跪求咋解决的？谢谢

MatthewDevelop, 2020-07-21:
打包为arr之后需要在自己的项目中添加依赖 implementation "com.serenegiant:common:4.1.1"

MatthewDevelop, 2020-07-21:
打包为aar之后需要在自己的项目中添加依赖 implementation "com.serenegiant:common:4.1.1"