jiangdongguo/AndroidUSBCamera #92: 点录制就报错

wasdzy111, 2018-12-20:
java.lang.NullPointerException: Attempt to invoke virtual method 'int android.media.MediaMuxer.addTrack(android.media.MediaFormat)' on a null object reference
        at com.serenegiant.usb.encoder.biz.Mp4MediaMuxer.addTrack(Mp4MediaMuxer.java:55)
        at com.serenegiant.usb.encoder.biz.H264EncodeConsumer.run(H264EncodeConsumer.java:174)

scJingP, 2018-12-25:
我的也是，你的解决了吗？
我的提示是 couldn't find a good color format for

wasdzy111, 2019-01-09:
最后把源码下载下来 使用可以运行；使用implementation 'com.github.jiangdongguo:AndroidUSBCamera:2.2.8'不得行，不知道为什么

117561271, 2019-08-28:
最新的源码编译后还是会有问题
BOARD=sdm660
CPU_ABI2=
HOST=compiler030
versionName=1.1.0.20190826
IS_TREBLE_ENABLED=true
SUPPORTED_64_BIT_ABIS=[Ljava.lang.String;@a37694d
CPU_ABI=arm64-v8a
PERMISSIONS_REVIEW_REQUIRED=false
IS_USERDEBUG=false
DISPLAY=OPM1.171019.011 release-keys
SUPPORTED_ABIS=[Ljava.lang.String;@a3b7c02
FINGERPRINT=vivo/PD1728/PD1728:8.1.0/OPM1.171019.011/compil01222023:user/release-keys
PRODUCT=PD1728
ID=OPM1.171019.011
TYPE=user
SERIAL=a753e09d
unauthorizedAppList=[pa@bcqn@n@cubabnyohzfubhfubh, pbz@n1qri@fqorapu, pbz@naqebzrqn@naqeborapu2, pbz@naghgh@NOrapuZnex, pbz@naghgh@NOrapuZnex@TY3, pbz@naghgh@PchZnfgreSerr, pbz@naghgh@orapuznex@shyy, pbz@naghgh@grfgre, pbz@nah@znva@zlnaqebvq, pbz@pnobqvqri@guerrqorapuznex, pbz@pbbyreznfgre@pchpbbyre@pbbyqbja@serr, pbz@pchvq@pch_m, pbz@pchvq@pch_m_pa, pbz@pchzbavgbe@rnfl, pbz@qvnakvabf@bcgvzvmre@qhcynl, pbz@qc@flfzbavgbe@ncc, pbz@svanyjver@nvqn64, pbz@shgherznex@qznaqebvq@nccyvpngvba, pbz@shgherznex@cpznex@naqebvq@orapuznex, pbz@uneqjner@pchpbbyre@pchk, pbz@yvkvnatqbat@zbovyrzbavgbe, pbz@yhqnfuv@orapuznex, pbz@yhqnfuv@orapuznexuq, pbz@atbguvrhdhnat@frgpchserr, pbz@cevzngrynof@trrxorapu, pbz@dhvpvap@iryynzb, pbz@graprag@ddcvzfrpher, pbz@irangbe@ncc@hefnsrpchzbavgbe, pbz@ivfx@fvzcyrfrgpch, syne2@qripurpx, cx@pbz@wnmor@naqebvqflfgrzvasb, fqt@fl@cresbeznaprgbby, mnhfna@mqrivprgrfg]
IS_CONTAINER=false
IS_ENG=false
DEVICE=PD1728
TIME=1548159786000
IS_USER=true
MODEL=vivo X21
MANUFACTURER=vivo
USER=compiler
versionCode=1
BRAND=vivo
SUPPORTED_32_BIT_ABIS=[Ljava.lang.String;@a2e0e4
HARDWARE=qcom
IS_DEBUGGABLE=false
BOOTLOADER=unknown
RADIO=unknown
TAG=Build
UNKNOWN=unknown
IS_EMULATOR=false
TAGS=release-keys
com.serenegiant.usb.encoder.biz.Mp4MediaMuxer.addTrack(Mp4MediaMuxer.java:60)
com.serenegiant.usb.encoder.biz.H264EncodeConsumer.run(H264EncodeConsumer.java:167)
exception：Attempt to invoke virtual method 'int android.media.MediaMuxer.addTrack(android.media.MediaFormat)' on a null object reference

mzc007, 2019-09-04:
一录制视频就闪退，报这两个错
E/H264EncodeConsumer: couldn't find a good color format for OMX.MTK.VIDEO.ENCODER.AVC / video/avc
E/H264EncodeConsumer: Unable to find an appropriate codec for video/avc