jiangdongguo/AndroidUSBCamera #97: Usb camera permission dialog closed!!

vigneshwaranvelmurugan, 2019-01-23:
java.lang.IllegalArgumentException: surface should be one of SurfaceHolder, Surface or SurfaceTexture: null
        at com.serenegiant.usb.common.AbstractUVCCameraHandler.startPreview(AbstractUVCCameraHandler.java:193)
        at com.serenegiant.usb.common.UVCCameraHandler.startPreview(UVCCameraHandler.java:123)
        at com.jiangdg.usbcamera.UVCCameraHelper.startPreview(UVCCameraHelper.java:312)
        at com.jiangdg.usbcamera.UVCCameraHelper$1$1.run(UVCCameraHelper.java:118)
        at java.lang.Thread.run(Thread.java:764)

jiangdongguo, 2019-02-19:
please try to add those codes in your AndroidMnifest.xml:

<uses-permission android:name="android.permission.USB_PERMISSION" />
<uses-feature
        android:name="android.hardware.usb.host"
        android:required="true" />

yeminnhtut, 2019-03-28:
It seems to be an issue with SDK versions. I have the same problem when I was trying with compileSdkVersion 28 and targetSdkVersion 28. I changed them to 25 and 22, respectively, just like in the sample app. It worked.

gxh-apologize, 2019-04-26:
I have the same exception accidentally . don't know why.

whuan1024, 2020-06-18:
遇到同样的异常，运行作者的demo app没有问题，但是把libusbcamera库移到自己的应用里之后，就报这个错误：java.lang.IllegalArgumentException: surface should be one of SurfaceHolder, Surface or SurfaceTexture: null

gradle里面
compileSdkVersion 28
targetSdkVersion 27
运行在rk3399板子上，android版本是9.0

KanwalKhan-20, 2021-04-28:
Facing the same problem, initially, the application is working fine but after few days it is giving the above-mentioned error.
Any solution to this problem?