jiangdongguo/AndroidUSBCamera #99: My app can not load native library any more

quocnhat7, 2019-02-18:
Hi,
my app already had some native libraries at "src\main\jniLibs\armeabi", my app run normally.
+ I can create another similar sample with your sample, it also run ok
+ I start to integrate your lib to my main app, I can build ok.
I start to run my app, at first start, my app will call some methods in MY native lib, it crash because my app can not load my native lib.
I create new folder armeabi-v7a and armeabi-v8a, then copy all my lib in "armeabi", also same, can not load my lib.
If I throw away your lib, my app can run normally again. So, can you tell me what the reason is which you think?
I think your library has special config for native lib, then, it makes my app does not point to correct native lib folder => app did not load my libs.

Thank you

quocnhat7, 2019-02-18:
I may know the reason.
In the log, I see "dlopen failed: my_lib.so" is 32-bit instead of 64-bit
=> so, your lib was built 64bit, my lib just 32bit.
Which version of your library was built 32bit? May I know and point to it?

quocnhat7, 2019-02-18:
I view logs, detail here [https://prnt.sc/mml22q](url)
So, your lib include 4 folders arm64-v8a, armeabi-v7a, armeabi, x86, but my app only include 1 lib folder (I dont need to build for other platform, just need "armeabi"). When running on device arm64-v8a, final lib folder will NOT include my lib, so my app can not load (this is common reason when integrating with other lib). I can not create new lib folder "arm64-v8a", because my lib is built with 32bit only :((.

So, can you instruct me how to let your lib only support for "armeabi" only (dont have other 3 folders)???
Another way I may think about is: I attach your lib as a module in my source, it will be ok.
But your code at github is not latest, can your share your latest code?

jiangdongguo, 2019-02-19:
Hi,modify your app gradle like this:

android {
     ...
    defaultConfig {
        ...
        // set supporting "armeabi"
        ndk {
            abiFilters "armeabi"
        }
    }
    ...
}

quocnhat7, 2019-02-22:
Thank you for help

closed 2019-02-22 by quocnhat7