jiangdongguo/AndroidUSBCamera #104: Crash if plug out my camera

quocnhat7, 2019-03-08:
Hi,
thank you for this lib, now I can connect to my external camera and get the video.
- I added your lib into my app. My app also has some other native libs to do something with video (libffmpeg, ...)
- After I plug the camera out, I can receive callback onDisConnectDev(), then onDettachDev().
- But 1-4 seconds later, I got crash in your lib, please see here [https://prnt.sc/mv0wxz](url)
- In method onDettachDev(), I already have the code:
+ mCameraHelper.closeCamera();
+ mCameraHelper.release();
=> If I disable these 2 functions, lib also got crash.

Note: if I run your sample app, there is no problem when I plug camera out. Perhaps there is something special will make lib got crash?

Please help. I am using version com.github.jiangdongguo:AndroidUSBCamera:2.2.8

quocnhat7, 2019-03-14:
Hi, can you help me :(

yu12mang, 2019-05-09:
i have same problem ,anyone can help us?

AsmodanPushPlay, 2019-06-13:
It would be nice to have more information about the problem. 

Debugging without USB is not that hard. I would reccomend an App called MatLog. Follow the instructions and you can view your Android log. It helps a lot and gives you a very clear list what the problem is and where it came from.

Debugging without any hint is very difficult.

quocnhat7, 2019-06-14:
@AsmodanPushPlay thank you, will try that lib.
@jiangdongguo we found problem. The native lib will encounter this crash on some devices running android 9 (not only because of android 9. I have 2 phones honor and xiaomi, 1 is ok, 1 got crash).
Our team rebuild the native lib from parent project, it is ok now. We will submit that update to parent lib project. Your lib will be safe also with this crash :D

mporetvivint, 2019-07-19:
I don't get a craash, but it stops executing at thread.mSync.wait(); It never moves on from this

quocnhat7, 2019-07-23:
> I don't get a craash, but it stops executing at thread.mSync.wait(); It never moves on from this

- If you run on other phone android 8, 7, .. is it ok?
- How can we know your problem if only info "thread.mSync.wait()" :3, where is it?
- Pls debug yourself first, at least you can know the reason