jiangdongguo/AndroidUSBCamera #116: Android 9.0 拔出摄像头，会闪退

AlwaysForwardFJJ, 2019-04-26:
Android 9.0 拔出摄像头，会闪退，报SIGSEGV(SEGV_MAPERR) 提示是so的错

1 #00 pc 00011c34 /data/app/jg.hczh-5I_xeBsE2BzjNZ-b7vDVgw==/lib/arm/libusb100.so [armeabi-v7a]
--
2 #01 pc 0000d200 /data/app/jg.hczh-5I_xeBsE2BzjNZ-b7vDVgw==/lib/arm/libusb100.so [armeabi-v7a]
3 #02 pc 0000cb60 /data/app/jg.hczh-5I_xeBsE2BzjNZ-b7vDVgw==/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+968) [armeabi-v7a]
4 #03 pc 0000d3cc /data/app/jg.hczh-5I_xeBsE2BzjNZ-b7vDVgw==/lib/arm/libusb100.so (libusb_handle_events+52) [armeabi-v7a]
5 #04 pc 00012c20 /data/app/jg.hczh-5I_xeBsE2BzjNZ-b7vDVgw==/lib/arm/libuvc.so (_uvc_handle_events+136) [armeabi-v7a]

pumatus, 2019-05-08:
plus

yu12mang, 2019-05-09:
我也遇到这个问题 请问解决了吗？

quocnhat7, 2019-06-09:
I got the same, crash android 9 Honor phone when plug camera out.
No problem on other phones with older android versions.

AsmodanPushPlay, 2019-06-13:
Have you been able to run the App on Android 9? I would be happy to deal with that problem in the future. At the moment I try to figure out why Android Pie is auch a big deal. I have no problems with other Android versions. Runs like a charm.

quocnhat7, 2019-06-14:
@AlwaysForwardFJJ @AsmodanPushPlay 
see my latest comment [https://github.com/jiangdongguo/AndroidUSBCamera/issues/104](url)

zhyang18, 2019-06-20:
plus

lbb1993, 2019-06-21:
我也遇到了同样的情况9.0拔线闪退

YiZhou213, 2019-07-06:
我也遇到了同样的情况拔线就会闪退

pumatus, 2019-08-10:
目前没有解决方案吗，我会一直关注此问题的

kylukuan, 2019-08-11:
有没有解决方案