jiangdongguo/AndroidUSBCamera #120: 麻烦问一下，编译APK失败报Error: Program type already present: com.jiangdg.libusbcamera.BuildConfig

WestbrookZero, 2019-05-09:
Error: Program type already present: com.jiangdg.libusbcamera.BuildConfig

jiangdongguo, 2019-05-14:
clean下试试

WestbrookZero, 2019-05-14:
已解决了，谢谢，我是不把compile改成implementation就好了，然后有个问题顺便想要麻烦问一下，很棘手一直都解决不了，红外摄像头的UVCCamera的640x260的分辨率怎么也显示不出来图像？

quocnhat7, 2019-06-03:
> 已解决了，谢谢，我是不把compile改成implementation就好了，然后有个问题顺便想要麻烦问一下，很棘手一直都解决不了，红外摄像头的UVCCamera的640x260的分辨率怎么也显示不出来图像，该如何解决？

This size is in "supported preview size" of your camera or not?
How about when you change the format?
mCameraHelper = UVCCameraHelper.getInstance();
mCameraHelper.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_MJPEG);

zhangze123456, 2019-06-09:
您好，我在项目中打包也存在这样的问题 项目移植后，可以显示外置USB摄像头的功能效果，可以安装install apk到手机，可是不能打包签名，报如下错误程序类型已存在：com.serenegiant.usb.CameraDialog $ 1，项目中有重复依赖。尝试改包名有问题，有没有遇到过的，请问，是怎么解决好的？谢谢
> 错误：程序类型已存在：com.jiangdg.libusbcamera.BuildConfig

> 错误：程序类型已存在：com.jiangdg.libusbcamera.BuildCon