jiangdongguo/AndroidUSBCamera #125: Still showing dialog to check permission

quocnhat7, 2019-06-03:
Hi,
in my app, I have code:  mCameraHelper.requestPermission(0)
+ I checked on checkbox to remember it next times
+ I granted permission.
+ I use external camera, then plugin it out and plug in again.
=> I still see the dialog to ask me approve again
My app info: compileSdk = 28, targetSdk = 26, buildTools = "28.0.3"
Please help.

RiponDas, 2019-06-09:
This lib sdk version is 25.3.1, if you upgrade this version the lib unable to take picture and other properties.
Set compileSdk = 25, targetSdk = 22, buildTools = "26.0.2"
and try again. Hope this work not showing dialog again.

quocnhat7, 2019-06-14:
> This lib sdk version is 25.3.1, if you upgrade this version the lib unable to take picture and other properties.
> Set compileSdk = 25, targetSdk = 22, buildTools = "26.0.2"
> and try again. Hope this work not showing dialog again.

thank, but it is not a reason.
I have 2 projects, one is target 27, one is target 22, they got the same result.
So, the problem is: **I will see "permission dialog" everytime I plug camera in (plug in first times, or plug out then plug in again)**

**At onAttachDev():**

//			mCameraHelper.requestPermission(0);
			UsbManager mUsbManager = (UsbManager)Global.getApplicationContext().getSystemService(Context.USB_SERVICE);
			boolean l = mUsbManager.hasPermission(device);
			boolean l2 = mCameraHelper.getUSBMonitor().hasPermission(device);
			boolean l3 = mCameraHelper.getUSBMonitor().requestPermission(device);
			logUsbCamera("PERMISSION at "+System.currentTimeMillis() +" => l1="+l+", l2="+l2+", l3="+l3);
=> L1, L3, L3 are FALSE when I plug in camera again.

quocnhat7, 2019-06-17:
@jiangdongguo can you help?

AsmodanPushPlay, 2019-06-19:
Try and add this line to your Manifest:

intent-filter
action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
intent-filter
meta-data android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
android:resource="@xml/device_filter"

Read more about it here:

http://blog.blecentral.com/2015/10/01/handling-usb-connections-in-android/

quocnhat7, 2019-06-26:
> intent-filter
> action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
> intent-filter
> meta-data android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
> android:resource="@xml/device_filter"

No, this is not the solution for my question.
But in your side, you did not encounter this issue? Pls read my first post above and follow my steps.

The code in your latest comment is for listening event from usb, while I am asking about usb permission after attached usb device. @AsmodanPushPlay

quocnhat7, 2019-06-28:
@jiangdongguo this bug is because USBManager.hasPermission(USBDevice device) always return FALSE when we plug the same camera in. Do you know the reason?

AsmodanPushPlay, 2019-06-30:
I don`t even ask for permission directly. I didn´t change the code. I just wrote those lines to my Manifest and that was it. Works great. I use 2 cameras and both have no problems with the permission.

quocnhat7, 2019-07-01:
> I don`t even ask for permission directly. I didn´t change the code. I just wrote those lines to my Manifest and that was it. Works great. I use 2 cameras and both have no problems with the permission.

@AsmodanPushPlay the new lib version 2.3 or lib 2.2.8?

AsmodanPushPlay, 2019-07-03:
@quocnhat7 

I run 2.2.8 until all bugs are found in 2.3. Updating is sometimes not the best you can do if something is new. Works with my main camera on Android 9 so I am cool with that but the second camera fails and can`t be found.

Ambika-Gowda, 2020-01-17:
Hi,
I tried the above library with USB Camera, it works and when first time I have connected the camera and app is launched first time, it asks for Null permission and then connects and gets the Camera Feed. It does not asks again for permission, when app is closed and launched again.

I have a customized camera, changed the class and subclass in device_filter.xml. When camera is connected and launched first time it asks for Null permission, n no of times and after some trials the camera will get launched.
I close the app and again open the app it again asks for Null permission, and after n no of trials the camera gets launched. The behavior of asking permission is random. Once I press the Null permission the following error appears (given below).

Library version: 2.2.8
 minSdkVersion 24
targetSdkVersion 28
.build:gradle: 3.1.2
Ndk version: 14b

Following error occurs:
019-03-14 03:07:29.182 5348-5376/ E/USBMonitor: could not connect to device /dev/bus/usb/001/023
2019-03-14 03:07:29.204 5348-5377/E/UsbManager: exception in UsbManager.openDevice
java.lang.IllegalArgumentException: device /dev/bus/usb/001/023 does not exist or is restricted
at android.os.Parcel.createException(Parcel.java:1946)
at android.os.Parcel.readException(Parcel.java:1910)
at android.os.Parcel.readException(Parcel.java:1860)
at android.hardware.usb.IUsbManager$Stub$Proxy.openDevice(IUsbManager.java:484)
at android.hardware.usb.UsbManager.openDevice(UsbManager.java:408)
at com.serenegiant.usb.USBMonitor$UsbControlBlock.(USBMonitor.java:1013)
at com.serenegiant.usb.USBMonitor$UsbControlBlock.clone(USBMonitor.java:1036)
at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:193)
at com.serenegiant.usb.common.AbstractUVCCameraHandler$CameraThread.handleOpen(AbstractUVCCameraHandler.java:510)
at com.serenegiant.usb.common.AbstractUVCCameraHandler.handleMessage(AbstractUVCCameraHandler.java:354)
at android.os.Handler.dispatchMessage(Handler.java:106)
at android.os.Looper.loop(Looper.java:193)
at com.serenegiant.usb.common.AbstractUVCCameraHandler$CameraThread.run(AbstractUVCCameraHandler.java:1036)
Caused by: android.os.RemoteException: Remote stack trace:
at com.android.server.usb.UsbHostManager.openDevice(UsbHostManager.java:455)
at com.android.server.usb.UsbService.openDevice(UsbService.java:263)
at android.hardware.usb.IUsbManager$Stub.onTransact(IUsbManager.java:70)
at android.os.Binder.execTransact(Binder.java:731)

mConnection = monitor.mUsbManager.openDevice(device);
at this point it returns Null, being the problem

@quocnhat7 @AsmodanPushPlay @jiangdongguo 
Please expecting any reply or solution!!

ColtonIdle, 2023-03-29:
usb devices lose permission when you unplug them. this behavior is expected