jiangdongguo/AndroidUSBCamera #127: Can I get camera data without showing preview on screen?

quocnhat7, 2019-06-04:
I would like to:
- plug external camera in
- send camera data to remote side
- dont show my local camera view
=> can and how I do this?
I see if I dont add camera-view in my xml or let it GONE, camera code will get crash.

jiangdongguo, 2019-06-13:
may be you can try to use TextureView.

quocnhat7, 2019-06-18:
Oh, original lib has new commit with my request, so some your next lib version may have it?
https://github.com/saki4510t/UVCCamera/pull/480

akshayah3, 2019-08-31:
@quocnhat7 Were you able to capture a photo without the preview?

akshayah3, 2019-08-31:
@jiangdongguo Is there a way to capture images periodically in a background service without any UI preview and surface involved? I am unable to capture an image without a surface

slobglob, 2019-10-08:
@quocnhat7 @akshayah3 
Did you manage to work something out?

slobglob, 2019-10-09:
@quocnhat7 @akshayah3 
Hey I was able to do so with setting the view outside of the screen and capture the data through this api:
`            mCameraHelper.setOnPreviewFrameListener(new AbstractUVCCameraHandler.OnPreViewResultListener() {
                @Override
                public void onPreviewResult(byte[] nv21Yuv) {
                }
            });
`
I'm also toggling the view visibility gone/visible on every initiation of the camera.
Not best solution but what I have so far, maybe I'll add a PR in the future to get the data without the view object at all.

Anyways, hope it helps someone.

slobglob, 2019-12-15:
Did anyone made any progress with this?