jiangdongguo/AndroidUSBCamera #128: AndroidUSBCamera module can't update gradle version to 28

RiponDas, 2019-06-04:
Dear viewers,
Please concern.
I need to update the gradle version to the project. When I try to update it to 28.0.0
The following error shows every time.

> Android dependency 'com.android.support:recyclerview-v7' has different version for the compile (25.3.1) and runtime (27.1.1) classpath. You should manually set the same version via DependencyResolution

Another way I try to use the project that making the project to a library and tat hen import to a new project where I got error with Xlint: deprecation,

What can I do to update gradle to 28?

quocnhat7, 2019-06-04:
I can use these info in my app, you can use it also:
buildTools = "28.0.3"
supportLibraryVersion = "27.1.1"
-----------
implementation 'com.android.support:recyclerview-v7:' + project.supportLibraryVersion

RiponDas, 2019-06-04:
@quocnhat7 
Thank you it works well and runs the project properly. But the problem is that image don't capture. Please tell me what can I do?
I need the image as a bitmap variable.

quocnhat7, 2019-06-06:
Can you post your code when capture picture (put code in a file)?
I don't try the code to capture and save picture yet. My app just get byte array from camera then process it.

RiponDas, 2019-06-08:
@jiangdongguo 
@quocnhat7 
Please help me on this issue. I have upgraded the dependencies like this. The project run but there is no image captured. No bitmap captured also.
Changes in Project build.gradle
`ext {
    supportLibraryVersion = '27.1.1'  // variable that can be referenced to keep support libs consistent
    commonLibVersion= '1.5.20'
    //versionBuildTool = '26.0.2'
    //versionCompiler = 25
    //versionTarget = 22
    //versionNameString = '1.0.0'
    javaSourceCompatibility = JavaVersion.VERSION_1_7
    javaTargetCompatibility = JavaVersion.VERSION_1_7
}
`
app build.gradle file is here. Here upgrading from 25.3.1 to 27.1.1
`apply plugin: 'com.android.application'

android {
    compileSdkVersion 27
    buildToolsVersion '27.0.3'
    defaultConfig {
        applicationId "com.jiangdg.usbcamera"
        minSdkVersion 16
        targetSdkVersion 27
        versionCode 1
        versionName "1.0"
        testInstrumentationRunner "android.support.test.runner.AndroidJUnitRunner"
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
}

dependencies {
    compile fileTree(include: ['*.jar'], dir: 'libs')
    androidTestCompile('com.android.support.test.espresso:espresso-core:2.2.2', {
        exclude group: 'com.android.support', module: 'support-annotations'
    })
    compile 'com.android.support:appcompat-v7:27.1.1'
    testCompile 'junit:junit:4.12'
    compile project(':libusbcamera')
    compile 'com.jakewharton:butterknife:8.8.1'
    annotationProcessor 'com.jakewharton:butterknife-compiler:8.8.1'
    implementation 'com.android.support:recyclerview-v7:' + project.supportLibraryVersion
}
`
Show this error message when try to take pic:
程序异常退出，即将重启...

Everything else is unchanged of this project.
N.B: I am unable to use this app in my app by following the instruction given.
Adding maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
        maven { url 'https://jitpack.io' }
in project build.gradle file and
implementation 'com.github.jiangdongguo:AndroidUSBCamera:2.2.8'

app build.gradle file.
In this case the project show 

> error: cannot access TaskStackBuilder
class file for android.support.v4.app.TaskStackBuilder not found

quocnhat7, 2019-06-09:
Hey, I don't understand.
At the end, you said "I am unable to use this app in my app by following the instruction given", that means you even can not add the lib into your own project. So, how can you capture the picture?

+ The sample project from this lib (https://github.com/jiangdongguo/AndroidUSBCamera), can you run it on your phone? If run ok, can you show camera preview, capture picture, ...?
+ You can zip your app here then I help to build it and check "can not capture", now I am confused.

RiponDas, 2019-06-09:
I tried multiple way.
Hey quocnhat7 ,
I change little in the sample project from this lib((https://github.com/jiangdongguo/AndroidUSBCamera).
My change is here.
`ext {
    supportLibraryVersion = '27.1.1'  // Change from 25.3.1 to 27.1.1
    commonLibVersion= '1.5.20'
    //versionBuildTool = '26.0.2'
    //versionCompiler = 25
    //versionTarget = 22
    //versionNameString = '1.0.0'
    javaSourceCompatibility = JavaVersion.VERSION_1_7
    javaTargetCompatibility = JavaVersion.VERSION_1_7
}`
update sdk to 27 in app gradle.build
` implementation 'com.android.support:recyclerview-v7:' + project.supportLibraryVersion`
when I try to take picture show a message: 程序异常退出，即将重启...
[AndroidUSBCamera-master.zip](https://github.com/jiangdongguo/AndroidUSBCamera/files/3269643/AndroidUSBCamera-master.zip)

RiponDas, 2019-06-10:
Dear quocnhat7,
Your project with sdk version=28 is working with camera capture or not. Would you share your project in zip. I upgrade to sdk 28 and it execute as usual but the problem one is that camera capture doesn't work. I am student of 3rd year. I am working on this for learning purpose only.

Thank you

Ripon das

jiangdongguo, 2019-06-13:
@RiponDas Hello, please show me your logs named 157789xxxx.txt in your root directory of sd card.

RiponDas, 2019-06-14:
com.jiangdg.usbcamera.view.USBCameraActivity$3$1.onCaptureResult(USBCameraActivity.java:194)
com.serenegiant.usb.common.AbstractUVCCameraHandler$CameraThread.saveYuv2Jpeg(AbstractUVCCameraHandler.java:848)
com.serenegiant.usb.common.AbstractUVCCameraHandler$CameraThread.access$400(AbstractUVCCameraHandler.java:391)
com.serenegiant.usb.common.AbstractUVCCameraHandler$CameraThread$3$1.run(AbstractUVCCameraHandler.java:813)
java.lang.Thread.run(Thread.java:818)
Abnormal：Attempt to invoke virtual method 'void android.widget.ImageView.setImageBitmap(android.graphics.Bitmap)' on a null object reference

Please consider log in my root directory. Don't show preview after removing butternife and don't capture photo.

jiangdongguo, 2019-06-15:
Hi @RiponDas ,

Have you ImageView been instanced?

RiponDas, 2019-06-15:
<?xml version="1.0" encoding="utf-8"?>
<RelativeLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:app="http://schemas.android.com/apk/res-auto"
    xmlns:tools="http://schemas.android.com/tools"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    android:background="#ff000000"
    tools:context=".view.USBCameraActivity"
    tools:ignore="MergeRootFrame">

    <FrameLayout
        android:layout_width="10dp"
        android:layout_height="5dp"
        android:layout_marginTop="-7dp">

        <com.serenegiant.usb.widget.UVCCameraTextureView
            android:id="@+id/camera_view"
            android:layout_width="match_parent"
            android:layout_height="wrap_content"
            android:layout_gravity="center" />

        <ImageView
            android:id="@+id/imageView"
            android:layout_width="match_parent"
            android:layout_height="match_parent" />

    </FrameLayout>
    <Button
        android:id="@+id/imageButton"
        android:layout_width="120dp"
        android:text="@string/app_name"
        android:layout_height="55dp"
        />

</RelativeLayout>

ImageView is an android widget to show image that captured. After removing butternife camera preview don't show.

quocnhat7, 2019-06-17:
@RiponDas 
+ all your phones got failure when capturing picture?
+ if you have only 1-2 phones, what are device models of them, model + android version?
Pls check app permission to save content in case app fail to save picture to sdcard.

jiangdongguo, 2019-06-18:
@RiponDas I have fixed your problem and please update the library to the new one.

AsmodanPushPlay, 2019-06-19:
@RiponDas 

Try this App: https://play.google.com/store/apps/details?id=com.pluscubed.matlog&hl=de

With this App it is possible find error stacktraces. You can read exactly which line of Code failed. You can Debug the App without a USB connection.

Saved me hours of work. if you are from China then you should read more about ADB without USB.

RiponDas, 2019-06-19:
@jiangdongguo 
@AsmodanPushPlay 
Thank you.
Actually when I upgrade sdk to 27 or 28 camera capture not working. This is my core problem that even unsolved.

closed 2019-06-22 by RiponDas

jessicanevesmachado, 2019-12-23:
> @jiangdongguo
> @AsmodanPushPlay
> Thank you.
> Actually when I upgrade sdk to 27 or 28 camera capture not working. This is my core problem that even unsolved.

Hello My Friend, do you can take picture now? if yes can you help me?