jiangdongguo/AndroidUSBCamera #129: 华为荣耀 pro10 连接，没有检测到usb

itkun2013, 2019-06-05:
试过https://www.jianshu.com/p/59eda43b89dc 这个方法 但是仍然不管用，请问你们怎么适配了
我在小米手机上没问题

quocnhat7, 2019-06-09:
Confirmed. Class subclass will be changed in android 9 devices :))
@itkun2013 in my source, I have to copy the whole source of UVCCameraHelper, then paste them into my own helper file. In that file, I change to my .XML file, I added some other class subclass values there.
Then I use my own helper file in the code to connect and work with external camera.

jiangdongguo, 2019-06-13:
可能是部分设备无法接收到USB设备插入广播，导致检测失败。我会尽快修复，看下能够有其他方案解决

jiangdongguo, 2019-06-13:
@quocnhat7 This is one of the failed reason on Android 9.0 and thank you for sharing it!

quocnhat7, 2019-06-14:
> @quocnhat7 This is one of the failed reason on Android 9.0 and thank you for sharing it out!

Yes, one of some reasons. Android 9 gives more troubles with system signals.
By the way, can you modify project source to let us (developers) add more devices into filter conditions?
From our tests (I tried 3-4 cameras until now), we really needs to update that XML file to recognize devices. Perhaps you provide a method to sample/app layer sending raw xml data to your lib?

Ex:
<usb-device class="239" subclass="2" />
	<usb-device class="14" subclass="9" />
	<usb-device class="2" subclass="0" />
	<usb-device class="6" subclass="-1" />
	<usb-device product-id="4836" vendor-id="9573" />
	<usb-device product-id="2229" vendor-id="1133" />
	<usb-device product-id="640" vendor-id="1409" />

pumatus, 2019-06-15:
您好，华为手机 nova型号 ，之前可以连接，随后过一段时间又不行了，不知道如何解决

jiangdongguo, 2019-06-17:
Hi @quocnhat7,
Thank you for your providing those useful information.About Android 9.0,the project will collect all of the failed connecting devices to a file named failed_device.txt and the file's path is root sd card/UsbCamera/failed_device.txt.

1159658610, 2019-06-17:
> 
> 
> 您好，华为手机 nova型号 ，之前可以连接，随后过一段时间又不行了，不知道如何解决

我也遇到这个问题,请问您解决了吗?

1159658610, 2019-06-18:
之前无法检测到9.0的华为设备,刚刚下载了作者最新的Project, 然后再华为9.0设备中手动创建了文件夹USBCamera/failed_devices.txt.然后程序就可以跑起来了.非常感谢作者.

pumatus, 2019-08-19:
> 之前无法检测到9.0的华为设备,刚刚下载了作者最新的Project, 然后再华为9.0设备中手动创建了文件夹USBCamera/failed_devices.txt.然后程序就可以跑起来了.非常感谢作者.

还需要手动创建么

1159658610, 2019-10-21:
嗯,我改成自动创建的了.