jiangdongguo/AndroidUSBCamera #131: 转换为库/Converting to library

RiponDas, 2019-06-10:
How do I use it in my project by converting a library? Please help.
如何通过转换库在我的项目中使用它？请帮忙。
By converting to the library I got the following error.
通过转换到库我得到以下错误。

error: cannot access IAspectRatioView class file for com.serenegiant.widget.IAspectRatioView not found |  
-- | --
error: cannot find symbol variable this |  
error: cannot find symbol variable super |  
error: cannot find symbol variable super |  
error: method does not override or implement a method from a supertype |  
error: method does not override or implement a method from a supertype |  
error: method does not override or implement a method from a supertype |

jiangdongguo, 2019-06-13:
Hi, 

In fact, The project has been modular and the module name is libusbcamera.If you want to use this library,may be you can realise it following these two steps：

- First, import libusbcamera as a module;
- second,add these configs in root gradle file:
allprojects {
    repositories {
        google()
        maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
        jcenter()
    }
}
ext {
    supportLibVersion = '25.3.1'  // variable that can be referenced to keep support libs consistent
    commonLibVersion= '1.5.20'
}

- Finally, learning USBCameraActivity.java and extract the core code.

Best regards！

RiponDas, 2019-06-14:
Same Error here.
What can I do to use this lib with sdk level 28?

RiponDas, 2019-06-15:
Error fixed. Thank you.

quocnhat7, 2019-06-17:
> Error fixed. Thank you.

pls close this report if it was fixed :))

closed 2019-06-19 by RiponDas