jiangdongguo/AndroidUSBCamera #132: Android Pie

AsmodanPushPlay, 2019-06-13:
Hi,

first I want to thank you for your great library. 

I would like to know if there is any way to run your library on Android Pie? I have no problems with Android 8, 7, 6 but when I try to run the App on Android Pie, nothing happens. The App won't ask for permissions, the App won't start and is unable to start the camera or a preview.

Is there any way to run the library on Android Pie? I would be very grateful for some help or hints.

jiangdongguo, 2019-06-13:
Hi, I have tested HUAWEI METE9 (Android 9.0) and my library worked ok.I suggest you that you can try to pull it out and connect it again.Of course,I also found that there is a little problem indeed on Android 9.0 and I do not know why so far.

AsmodanPushPlay, 2019-06-15:
Hi, i would like to thank you for your prompt response and assistance with my issue. 

I ran some tests and I think I might know what is wrong.

1. onAttach is being triggered but mCameraHelper.getUsbDeviceCount() == 0

2. I think the problem is that Android Pie is unable to detect the camera.

3. I used Google translator and I think this https://www.jianshu.com/p/59eda43b89dc
might be the answer to the question.

4. I will rewrite some classes and test if the solution works for me.

jiangdongguo, 2019-06-17:
@AsmodanPushPlay Android Pie can not detect the camera in time sometimes is a big trouble.

quocnhat7, 2019-06-17:
> @AsmodanPushPlay Android Pie can not detect the camera in time sometimes is a big trouble.

From this "onAttach is being triggered but mCameraHelper.getUsbDeviceCount() == 0"
=> 100% because your device_filter.xml not include that camera class+subclass, or device/vendor id

jiangdongguo, 2019-06-18:
@quocnhat7 That's right.In the new version,I provide a method of collecting those unregistering device by a file named failed_deivces.txt which is located in the directory /storage/.../0/UsbCamera.

AsmodanPushPlay, 2019-06-18:
Hi, thank you all for your support.

I created a custom UVCCameraHelper-Class and created a new devicefilter.xml File. Works great on Android 6, 7 and 8. Fails again on Android Pie.

I updated the library to the newest Version and had the problem, that my minSdkVersion needs to be 27, which I don`t want by now.

I am not shure if the App creates the problem, or my smartphone or the UVC-Camera.

I will put this issue aside for now and will deal with it in the future.

There is more work to do on other parts which are more fun. Dealing with Errors and Exceptions is compared to Android Pie fun and easy. Unknown problems are hard to deal with.

I tested the App on a Galaxy S8, Experia Z3 Compact and Galaxy S7 Edge.

Galaxy S8 worked with Android 8, S7 worked with Android 7, Experia Z3 worked with Android 6, failed on the Galaxy S8 with Android Pie.

I will try a Galaxy S10 with Android Pie in the future but for now I am done with Android Pie. I will keep you updated as soon as I know what the problem is.

Still a great library. And I still don`t like Android Pie.

AsmodanPushPlay, 2019-06-18:
@quocnhat7 - 100% because your device_filter.xml not include that camera class+subclass, or device/vendor id

I created a custom device_filter.xml and pointed to it with the correct R.xml.device_filter from my custom Helper Class. 

I used the VendorID and ProductID of the camera. All Android Versions work but Android Pie fails.

AsmodanPushPlay, 2019-06-18:
@jiangdongguo 

Hi, I did some tests and realized that Android Pie is probebly unable to get the Class/Subcalss of some Cameras. 

I wrote some code to get the Class/Subclass and it deliverd Class = 0 / Subclass = 0. First I thought my code is not working because on windows I get Class = 14 / Subclass = 3.

Then I registered the device in my Manifest by 

<intent-filter>
                <action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" />
            </intent-filter>
            <meta-data android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
                android:resource="@xml/device_filter" />

and 

<usb-device class="0" subclass="0"/>  in my device.xml file.

and Android registered the Camera. I am still not able do get a preview, because I probably have to rewrite the Helper-Class.

So the problem with Android Pie seems to be that it returns Class = 0 and Subclass = 0.

quocnhat7, 2019-06-19:
@AsmodanPushPlay more infor for you.
- I also fix that problem by creating another helper class, copy all orginal source, only use my own device_filter.xml with some more lines.
- A note for you: you have to add 2 lines with class/subclass value for your 1 device. Because on some kind of cameras:
+ In android 6/7/8, device.toString() return class=N, subclass = M
+ But same camera, in android 9, device.toString() return class = X, subclass = Y. This was reported by other people in another issue report.
- Another note from me: you also need to change JAVA code, to let your custom filter works
+ Go to your help java class
+ Find method getUsbDeviceList(), change it to 
`public List<UsbDevice> getUsbDeviceList() {
		List<DeviceFilter> deviceFilters = DeviceFilter
				.getDeviceFilters(mActivity.getApplicationContext(), R.xml.device_filteryishu);
		if (mUSBMonitor == null || deviceFilters == null)
			return null;
		**return mUSBMonitor.getDeviceList(deviceFilters);**
	}`
=> should not return "return mUSBMonitor.getDeviceList(deviceFilters.get(0));" in original source.

AsmodanPushPlay, 2019-06-19:
@quocnhat7

Hi, and thank you for the very useful information.

1. I created the custom defice_filter.xml and added:

usb-device class="0" subclass="0" --Unknown Devices--

2. I wrote a new UVCCameraHelper Class and used your custom getUSBDeviceList() method.

3. I also pointed to the correct R.xml.device_filter

4. I then pointed from my activity to the custom UVCCameraHelper Class.

And it works just fine.

I won`t close this issue so others can read it before they open a new one.

CONCLUSION: Android Pie sometimes delivers Class = 0 and Subclass = 0 with some cameras. 

Thank you very much for your help with this.

closed 2019-06-19 by AsmodanPushPlay