jiangdongguo/AndroidUSBCamera #137: 2.3.0 Release not build

RiponDas, 2019-06-19:
Release 2.3.0 doesn't build in android studio after opening shows this error message. It occurs due to compilation error for kotlin may be.

Please don't feel annoyed for the rapid question.

Thanks.....@jiangdongguo

quocnhat7, 2019-06-20:
You should clone sample app, then link to latest lib 2.3.0. If not work, it might be because of lib.
If ok, so it is because of your app dependencies.

If you link back to old lib 2.2.8, is it ok?

RiponDas, 2019-06-20:
No I have cloned this new lib 2.3.0 but show the error.

Content is not allowed in prolog.

RiponDas, 2019-06-23:
@jiangdongguo 
Please help me with the issue I faced. 

ERROR: Content is not allowed in prolog.

I can't resolve this problem until now.  Please help me.

closed 2019-06-24 by RiponDas