jiangdongguo/AndroidUSBCamera #138: Failure to recognize the camera after upgrading to 2.3.0

sava72, 2019-06-22:
Hello,

my USB camera was properly recognized till the previous version, I just attempted to install 2.3.0 and I've got a camera failure message. My phone is a Samsung Galaxy S9 with Android 9.

Thanks in advance for any tips.

Marco

AsmodanPushPlay, 2019-06-23:
Hi, 

first i would check if the device_list.xml has changed. Is your Class, Subclass still registered or did the update also effect the device_list.xml.

I fixed the problem with 2.2.8 version so I didn`t update to 2.3.0. Change something sometimes means lose everithing.

sava72, 2019-06-23:
Hi,

thanks for getting back to me. Actually I installed the newer version from scratch on another machine, and then I found out my camera was no longer working. 

I'll make a comparison between the 2 device_list.xml then.

Best,

Marco

RiponDas, 2019-06-23:
Hey @sava72 
I have encountered the same problem. If you fixed it please let me know.

AsmodanPushPlay, 2019-06-24:
First you should register an Intent Filter like this in your Manifest:

intent-filter>
action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" />
/intent-filter>
meta-data android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
android:resource="@xml/device_filter" />

If you plug in your camera, you should get a dialog from your App. If you don`t get the dialog, then your divice_filter. xml has the wrong Class/Subclass.

If you get the dialog and your App still can`t show you a preview then there is something wrong with our code or the library.

Sometimes the camera shows no preview because you set the wrong preview width and hight. There is a lot what can cause problems.

I am not the creator of this library so I can`t tell what could be wrong if the library is unable to detect a camera. Specially on Android pie.

sava72, 2019-06-24:
Hi again,

actually a dialog now pops up (it asks for automatically running the camera when plugged-in), but after saying yes, there's an "unknown error", it states to check the log file within the SD card to find out more. You have to forgive me but there's something still unclear to me: I can't find any device_filter.xml within the project, where should I look?

Thanks!

Marco

AsmodanPushPlay, 2019-06-25:
Hi, 

the device_filter.xml is part of the library and can be found here on Github inside the project code. If you want to declare your own device_list.xml then you need to rewrite the UVCHelper Class and point to your custom device_filter.xml file.

Now that we know that the device_filter.xml is ok we need to dig a bit deeper.

You need access to you error logs in order to find the problem. I use a app called MatLog because debugging by USB isn't possible if your camera is connected.

If you are able to post your log, it is much more likely to find the source of the problem.

quocnhat7, 2019-06-25:
> You need access to you error logs in order to find the problem. I use a app called MatLog because debugging by USB isn't possible if your camera is connected.

@AsmodanPushPlay 
You can use WIFI DEBUG feature of android studio.
- First, plug cable to connect your phone and Android studio
- Second, make sure your phone and PC are using sam wifi
- Run adb -d tcpip 7777
- Run adb connect phone_local_ip:7777
- Plug the cable out
Now you can view log in AS while you are plugging external camera in.

sava72, 2019-06-25:
Hi, 

attached the log, do you see anything meaningful?

Thanks,

Marco
[logs-2019-06-25-18-42-44.zip](https://github.com/jiangdongguo/AndroidUSBCamera/files/3326396/logs-2019-06-25-18-42-44.zip)

RiponDas, 2019-06-28:
@jiangdongguo 
@quocnhat7 
@AsmodanPushPlay 
The camera shows preview but can't take pictures.
Shows the error:
java.io.FileNotFoundException: /storage/emulated/0/USBCamera/failed_devices.txt: open failed: ENOENT (No such file or directory)
06-28 15:01:50.936 19787-19787/? E/Zygote: v2
06-28 15:01:50.936 19787-19787/? E/Zygote: accessInfo : 0
06-28 15:02:49.596 19787-19787/com.jiangdg.usbcamera E/MotionRecognitionManager: mSContextService = null
06-28 15:02:49.596 19787-19787/com.jiangdg.usbcamera E/MotionRecognitionManager: motionService = null
Here is my logcat: 
[LogCat.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/3338429/LogCat.txt)

AsmodanPushPlay, 2019-06-30:
Hi,

1. @RiponDas do you have permission to read or write your storage?

2. I would send the LogCat.txt to the creator of the library. I think he can help you out.

I didn`t write the library so I can´t help out if something does not work inside the library. jiangdongguo knows more than I do.

RiponDas, 2019-06-30:
Dear @AsmodanPushPlay 
Please try to send the logcat to the author. 
Here some acknowledgment:
1. I have permitted for reading or write in storage.
2. Show the dialog box for accepting usb camera. Then also shows the camera preview.
3. But there is no folder created named USBCamera in the storage and no picture stored.
4. In the logcat a Debug shows my picture stored in the following location 
I/Debug: save path：/storage/emulated/0/USBCamera/images/1561746702268.jpg
5.Also a waring shows:
java.io.FileNotFoundException: /storage/emulated/0/USBCamera/failed_devices.txt: open failed: ENOENT (No such file or directory)

Regards..

quocnhat7, 2019-07-01:
@RiponDas pls zip your project and send here, I will help you debug it.

RiponDas, 2019-07-01:
@quocnhat7 
I just edit this line as follows and camera preview starts to show. But no image captured yet.
return mUSBMonitor.getDeviceList(deviceFilters.get(0)); in UVCCamerahelper.java

Here is the project zip file. Link share for the excess size of the file.
https://mega.nz/#!Q7Z3BSiY!KxHLzSTNX9uTYIBORMPC7BnzPiu8K-Rl9bqfBrfjgiM

quocnhat7, 2019-07-03:
@RiponDas you should debug project to know where got bug.
The reason of bug is: CAPTURE/RECORDING will save image/video to sdcard, but @jiangdongguo forgot to create folder before saving data into it.
Fix it by: create folder if missing, before writing data there
https://mega.nz/#!3r4kSagY!XGccfzX1z8IsdAU3Zo759ByxEsGs3LEifNLrhK838Rc

quocnhat7, 2019-07-03:
You can see my 3 methods in MyApplication.java. In the Activity, just use these 3 methods when you want to choose folder destination.

RiponDas, 2019-07-03:
@quocnhat7 
I unable to use these 3 function in MyApplication.java and still I face the same problem. It would be helpfull for me if you use thes 3 function in destination folder choosing.

AsmodanPushPlay, 2019-07-03:
@RiponDas 

Why don`t you write your own custom code to store your pictures? I use my own directory and methods because I need to. You should not count to much on libraries. The best way is to be as independent as possible from external code.

quocnhat7, 2019-07-09:
> @quocnhat7
> I unable to use these 3 function in MyApplication.java and still I face the same problem. It would be helpfull for me if you use thes 3 function in destination folder choosing.

So, you please debug yourself. As I already modified in **your source code project**, and it works for sure (your original source does not work, I modified and work ok).
You should read my comment above to understand the reason.

quocnhat7, 2019-07-23:
@RiponDas it is ok now?

ejscribner, 2020-07-16:
I am also getting the error: `W/System.err: java.io.FileNotFoundException: /storage/emulated/0/USBCamera/failed_devices.txt (Permission denied)` but I believe my permissions are set up correctly. Has anyone found a solution to this issue?

AsmodanPushPlay, 2020-07-16:
How can you believe that your permissions are correct?

1. Create a TEST.txt File

2. Read from the TEST.txt File

You should test your code first.

quocnhat7, 2020-07-18:
> I am also getting the error: `W/System.err: java.io.FileNotFoundException: /storage/emulated/0/USBCamera/failed_devices.txt (Permission denied)` but I believe my permissions are set up correctly. Has anyone found a solution to this issue?

Now, open the Settings of your phone, find that sample app and check the Storage permission is checked or unchecked. 
+ if it is checked, it is strange, you should post the code here
+ If unchecked, check it and try again (if it is unchecked, for sure that is the code to request permission not work, need to check your java code again)