jiangdongguo/AndroidUSBCamera #141: 预览界面全屏设置，如何满足不同机型的适配

zhangze123456, 2019-07-01:
mCameraHelper.setDefaultPreviewSize(1280,720);此参数应该如何调节，
如何设置预览尺寸大小，现在屏幕是水平居中，没有充满屏幕，如何设置可以使预览界面全屏，适用不同的机型，请问大家有没有解决办法，谢谢大家

quocnhat7, 2019-07-01:
> mCameraHelper.setDefaultPreviewSize(1280,720);此参数应该如何调节，
> 如何设置预览尺寸大小，现在屏幕是水平居中，没有充满屏幕，如何设置可以使预览界面全屏，适用不同的机型，请问大家有没有解决办法，谢谢大家

previewSize is the size which camera use to open and capture picture/video. You can try to change it from 1280,720 to 640,480 to see the different in quality of preview picture.

zhangze123456, 2019-07-02:
设置了setDefaultPreviewSize中参数非1280 720之后，1920 1080后界面摆平，请问修改哪里可以达到，16:9 4:3 18:9的预览界面

quocnhat7, 2019-07-03:
com.serenegiant.usb.widget.UVCCameraTextureView
=> can edit the way to show video on screen here.
But a note that: keeping picture aspect ratio is a mandatory requirement for any app displaying camera preview. @zhangze123456