jiangdongguo/AndroidUSBCamera #143: java.lang.IllegalArgumentException: surface should be one of SurfaceHolder, Surface or SurfaceTexture: null

gxh-apologize, 2019-07-09:
异常：
`07-07 17:30:28.171 E/AndroidRuntime( 1462): java.lang.IllegalArgumentException: surface should be one of SurfaceHolder, Surface or SurfaceTexture: null
07-07 17:30:28.171 E/AndroidRuntime( 1462): 	at com.serenegiant.usb.common.AbstractUVCCameraHandler.startPreview(AbstractUVCCameraHandler.java:193)
07-07 17:30:28.171 E/AndroidRuntime( 1462): 	at com.serenegiant.usb.common.UVCCameraHandler.startPreview(UVCCameraHandler.java:123)
07-07 17:30:28.171 E/AndroidRuntime( 1462): 	at com.jiangdg.usbcamera.UVCCameraHelper.startPreview(UVCCameraHelper.java:312)
07-07 17:30:28.171 E/AndroidRuntime( 1462): 	at com.jiangdg.usbcamera.UVCCameraHelper$1$1.run(UVCCameraHelper.java:118)
07-07 17:30:28.171 E/AndroidRuntime( 1462): 	at java.lang.Thread.run(Thread.java:818)`

//使用的版本
implementation 'com.github.jiangdongguo:AndroidUSBCamera:2.2.8'

rk3288  Android5.1.1   这个异常是有时候会发生，并不是100%。我们的项目是开机自启动的，开机后有时候会报这个异常。望回复。谢谢

hyunahOh, 2020-01-07:
Did you solve it?

ankittecholution, 2020-01-26:
Is there any solution to this problem???

RAZA-ABBBASI, 2020-06-22:
Mostly it happens if we try to register USB devices while the layout containing the textureview isn't active/visible. i.e., in viewpager and fragment combination I was registering USB in the start function which is executed before we actually move to the fragment ( if fragment isn't the first fragment) then I was getting same error. I did fix that by moving register code to onResume with a chec.

gxh-apologize, 2020-06-30:
后面我们使用的原生camera来预览的，效果也很好，usb摄像头也可以用原生camera，这个问题并未深究。

VishnuTB, 2020-08-12:
Any update on this?

VishnuTB, 2020-08-13:
@jiangdongguo  any update on this?
I'm using _com.github.jiangdongguo:AndroidUSBCamera:2.3.1_