jiangdongguo/AndroidUSBCamera #161: sorry,open camera failed

1025342523, 2019-08-16:
java.io.IOException: open failed: ENOENT (No such file or directory)
12-28 22:12:46.571 3543-4143/com.jiangdg.usbcamera W/System.err:     at java.io.File.createNewFile(File.java:939)
12-28 22:12:46.572 3543-4143/com.jiangdg.usbcamera W/System.err:     at com.serenegiant.usb.USBMonitor.getDeviceList(USBMonitor.java:319)
12-28 22:12:46.572 3543-4143/com.jiangdg.usbcamera W/System.err:     at com.serenegiant.usb.USBMonitor.getDeviceList(USBMonitor.java:300)
12-28 22:12:46.572 3543-4143/com.jiangdg.usbcamera W/System.err:     at com.serenegiant.usb.USBMonitor$2.run(USBMonitor.java:579)
12-28 22:12:46.572 3543-4143/com.jiangdg.usbcamera W/System.err:     at android.os.Handler.handleCallback(Handler.java:739)
12-28 22:12:46.572 3543-4143/com.jiangdg.usbcamera W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
12-28 22:12:46.572 3543-4143/com.jiangdg.usbcamera W/System.err:     at android.os.Looper.loop(Looper.java:148)
12-28 22:12:46.572 3543-4143/com.jiangdg.usbcamera W/System.err:     at android.os.HandlerThread.run(HandlerThread.java:61)
12-28 22:12:46.572 3543-4143/com.jiangdg.usbcamera W/System.err: Caused by: android.system.ErrnoException: open failed: ENOENT (No such file or directory)
12-28 22:12:46.572 3543-4143/com.jiangdg.usbcamera W/System.err:     at libcore.io.Posix.open(Native Method)
12-28 22:12:46.573 3543-4143/com.jiangdg.usbcamera W/System.err:     at libcore.io.BlockGuardOs.open(BlockGuardOs.java:186)
12-28 22:12:46.573 3543-4143/com.jiangdg.usbcamera W/System.err:     at java.io.File.createNewFile(File.java:932)
12-28 22:12:46.573 3543-4143/com.jiangdg.usbcamera W/System.err: 	... 7 more
12-28 22:12:46.575 3543-4143/com.jiangdg.usbcamera W/System.err: java.io.FileNotFoundException: /storage/emulated/0/USBCamera/failed_devices.txt: open failed: ENOENT (No such file or directory)
12-28 22:12:46.575 3543-4143/com.jiangdg.usbcamera W/System.err:     at libcore.io.IoBridge.open(IoBridge.java:452)
12-28 22:12:46.575 3543-4143/com.jiangdg.usbcamera W/System.err:     at java.io.FileOutputStream.<init>(FileOutputStream.java:87)
12-28 22:12:46.575 3543-4143/com.jiangdg.usbcamera W/System.err:     at java.io.FileWriter.<init>(FileWriter.java:58)
12-28 22:12:46.576 3543-4143/com.jiangdg.usbcamera W/System.err:     at com.serenegiant.usb.USBMonitor.getDeviceList(USBMonitor.java:328)
12-28 22:12:46.576 3543-4143/com.jiangdg.usbcamera W/System.err:     at com.serenegiant.usb.USBMonitor.getDeviceList(USBMonitor.java:300)
12-28 22:12:46.576 3543-4143/com.jiangdg.usbcamera W/System.err:     at com.serenegiant.usb.USBMonitor$2.run(USBMonitor.java:579)
12-28 22:12:46.576 3543-4143/com.jiangdg.usbcamera W/System.err:     at android.os.Handler.handleCallback(Handler.java:739)
12-28 22:12:46.576 3543-4143/com.jiangdg.usbcamera W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
12-28 22:12:46.576 3543-4143/com.jiangdg.usbcamera W/System.err:     at android.os.Looper.loop(Looper.java:148)
12-28 22:12:46.576 3543-4143/com.jiangdg.usbcamera W/System.err:     at android.os.HandlerThread.run(HandlerThread.java:61)

String fileName = Environment.getExternalStorageDirectory()+ "/USBCamera/failed_devices.txt";

		File logFile = new File(fileName);
		if(! logFile.exists()) {
			try {
				logFile.createNewFile();
			} catch (IOException e) {
				e.printStackTrace();
			}
		}
		FileWriter fw = null;
		PrintWriter pw = null;
		final List<UsbDevice> result = new ArrayList<UsbDevice>();
		try {
                       // 报错的地方
			fw = new FileWriter(logFile, true);

jiangdongguo, 2019-08-26:
have fixed,please have a try.

1025342523, 2019-09-10:
已解决

closed 2019-09-10 by 1025342523