jiangdongguo/AndroidUSBCamera #166: 拍照没生成图片

moyaying, 2019-08-29:
摄像头能正常显示，但是拍照没生成图片。
我把minSdkVersion 调成了19.
mCameraHelper?.setOnPreviewFrameListener {
                Timber.i("on usb camera preview result")
            }

这句话一直没有回调。

jiangdongguo, 2019-09-18:
这个可能需要跟踪下代码，我的是没问题，但是不同的摄像头设备还是有差异的。

ailala1993929, 2020-01-08:
摄像头能正常预览，但是拍照没生成图片，onCaptureResult一直没回调。
mCameraHelper.capturePicture(picPath, new AbstractUVCCameraHandler.OnCaptureListener() {
                    @Override
                    public void onCaptureResult(String path) {
                        Log.i(TAG,"save path：" + path);
                        Toast.makeText(USBCameraActivity.this,"AAAApath:"+path,Toast.LENGTH_SHORT ).show();
                    }
                });

ailala1993929, 2020-01-08:
case MSG_CAPTURE_STILL:
				thread.handleCaptureStill((String)msg.obj);
                thread.handleStillPicture((String) msg.obj);
                break;
// 捕获静态图片
        public void handleCaptureStill(final String path) {
            if (DEBUG) Log.v(TAG_THREAD, "handleCaptureStill:");
            final Activity parent = mWeakParent.get();
            if (parent == null) return;
//			mSoundPool.play(mSoundId, 0.2f, 0.2f, 0, 0, 1.0f);	// play shutter sound
            try {
                final Bitmap bitmap = mWeakCameraView.get().captureStillImage(mWidth, mHeight);
                // get buffered output stream for saving a captured still image as a file on external storage.
                // the file name is came from current time.
                // You should use extension name as same as CompressFormat when calling Bitmap#compress.
                final File outputFile = TextUtils.isEmpty(path)
                        ? MediaMuxerWrapper.getCaptureFile(Environment.DIRECTORY_DCIM, ".jpg")
                        : new File(path);
                final BufferedOutputStream os = new BufferedOutputStream(new FileOutputStream(outputFile));
                try {
                    try {
                        bitmap.compress(Bitmap.CompressFormat.JPEG, 100, os);
                        os.flush();
                        mHandler.sendMessage(mHandler.obtainMessage(MSG_MEDIA_UPDATE, outputFile.getPath()));
                    } catch (final IOException e) {
                    }
                } finally {
                    os.close();
                }
                if (mCaptureListener != null) {
                    mCaptureListener.onCaptureResult(path);
                }
            } catch (final Exception e) {
                callOnError(e);
            }
        }
应该是调用的这个函数吧？

ailala1993929, 2020-01-08:
找到问题了：
case MSG_CAPTURE_STILL:
thread.handleCaptureStill((String)msg.obj);
//thread.handleStillPicture((String) msg.obj);
break;
这里的函数调用需要调用上面源码注释掉的函数，thread.handleCaptureStill；
还有一个点就是图片保存路径的问题：String picPath = UVCCameraHelper.ROOT_PATH + MyApplication.DIRECTORY_NAME +"/images/"+ System.currentTimeMillis()
+ UVCCameraHelper.SUFFIX_JPEG;
这里的images还有外面的USBCamera文件夹需要手动去创建，是bug，所以拍照图片未保存到指定路径。