jiangdongguo/AndroidUSBCamera #170: 部分摄像头会卡帧，花帧

wsl1015310333, 2019-09-04:
有一些摄像头能正常显示，部分摄像头链接后大几率出现花帧，掉帧卡顿，app卡顿很严重。用手当住摄像头就好了

jiangdongguo, 2019-09-18:
是不是高分辨率摄像头，然后选择YUV格式采集？建议选择MJPEG试试

wsl1015310333, 2019-09-26:
public static final int PIXEL_FORMAT_RAW = 0;
public static final int PIXEL_FORMAT_YUV = 1;
public static final int PIXEL_FORMAT_RGB565 = 2;
public static final int PIXEL_FORMAT_RGBX = 3;
public static final int PIXEL_FORMAT_YUV420SP = 4;
public static final int PIXEL_FORMAT_NV21 = 5;	
第几个模式 是 MJPEG 看了源码没找到 。

wsl1015310333, 2019-10-25:
private int mFrameFormat = FRAME_FORMAT_MJPEG;
        mCameraHelper.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_MJPEG);
从新设置分辨时候调用这个
 mCameraHandler = UVCCameraHandler.createHandler(mActivity,mCamView, 2,
                previewWidth, previewHeight, mFrameFormat);
我看你默认就是设置这个参数

wsl1015310333, 2019-10-25:
我看你默认项目就是mjpeg

wsl1015310333, 2019-10-25:
他的值是1 Yuv是0

GQii, 2019-12-23:
@wsl1015310333 楼主，你这个花屏、掉帧的问题解决了吗？是如何解决的？

KingMing2017, 2020-12-01:
> @wsl1015310333 楼主，你这个花屏、掉帧的问题解决了吗？是如何解决的？

同问，解决了吗？

vc815, 2022-05-30:
> 有一些摄像头能正常显示，部分摄像头链接后大几率出现花帧，掉帧卡顿，app卡顿很严重。用手当住摄像头就好了

我就遇到了，手一遮挡，画面就正常了，不遮挡就一直是灰屏，卡顿，，解决了么？
感觉是高分率出现这样情况概率高大很多