jiangdongguo/AndroidUSBCamera #172: startPusher 录像失败

arthas2010, 2019-09-11:
错误信息：
SUPPORTED_64_BIT_ABIS=[Ljava.lang.String;@e376cfa
versionCode=1
BOARD=msm8909
BOOTLOADER=unknown
TYPE=user
ID=LMY47V
TIME=1561977573000
BRAND=qcom
TAG=Build
SERIAL=ba76b1f
HARDWARE=qcom
SUPPORTED_ABIS=[Ljava.lang.String;@3429c2ab
CPU_ABI=armeabi-v7a
RADIO=unknown
IS_DEBUGGABLE=false
MANUFACTURER=unknown
SUPPORTED_32_BIT_ABIS=[Ljava.lang.String;@26092325
TAGS=test-keys
CPU_ABI2=armeabi
UNKNOWN=unknown
USER=yuan
FINGERPRINT=qcom/msm8909/msm8909:5.1.1/LMY47V/yuan07011838:user/test-keys
HOST=yuan-S2400SC
PRODUCT=msm8909
versionName=1.1.0.20190826
DISPLAY=5.1.1.5
MODEL=msm8909
DEVICE=msm8909
com.serenegiant.usb.encoder.biz.Mp4MediaMuxer.addTrack(Mp4MediaMuxer.java:60)
com.serenegiant.usb.encoder.biz.H264EncodeConsumer.run(H264EncodeConsumer.java:165)
exceptionï¼Attempt to invoke virtual method 'int android.media.MediaMuxer.addTrack(android.media.MediaFormat)' on a null object reference

jiangdongguo, 2019-09-18:
空指针，换台手机试试？

arthas2010, 2019-09-20:
> 空指针，换台手机试试？

解决了，源码中又一次BUG，录像文件如果不存在就会报空指针

Hfengxiang, 2019-09-23:
怎么解决的啊，兄弟!!

zhoujielun370, 2019-11-21:
> 怎么解决的啊，兄弟!!

其实他的拍照和录制都没有问题，就是起初写路径有问题，有许多机子是不允许这样直接创建路径文件，必须是先new file  再 file.mkdirs(); 外面嵌套几层文件夹，必须通过mkdirs()逐步创建。