jiangdongguo/AndroidUSBCamera #173: sometimes the screen is gray in huawei mate10 while is preview

zhancaihai, 2019-09-11:
sometimes the screen is gray in huawei mate10 while is preview

jiangdongguo, 2019-09-18:
sometimes?when?

quocnhat7, 2019-09-20:
I think it is because of camera. Camera has auto focus feature, it will give a little gray, then light.
If camera preview is full gray, can not display video then, that is another bug => pls capture screenshot

zhancaihai, 2019-09-23:
就是在mate10上预览的时候，预览一段时间后，屏幕变成灰色的，偶尔会出现一点画面，退出去，重新进来就好了

zhancaihai, 2019-09-23:
不知道是不是fps变低了导致的

quocnhat7, 2019-09-26:
Sorry I dont really understand the reason, it is better to give us a video.
My app also is used in many phones and many cameras but I dont see many problems with video quality.

heaveninfosoft, 2020-10-14:
https://drive.google.com/file/d/1TVjQUHuTqXqg9iIAnmvXtwFsvvqTgoKU/view?usp=sharing

heaveninfosoft, 2020-10-14:
@quocnhat7 urgent please

quocnhat7, 2020-10-15:
- try another phone
- try another camera
=> give conclusion: problem caused by camera or phone. I think it is because of camera.
---
Then, use camera Helper to adjust camera encoder setting (codec, camera resolution such as VGA, HD, ...) => you might see which case can give good video.
=> you try more tests to analyze

heaveninfosoft, 2020-10-15:
@quocnhat7 it is related to because of high resolution camera?
i am using samsung 9 plus with 1080p camera and right now resolution is default resolution 640*480

quocnhat7, 2020-10-15:
When using lib external camera, you needs to open camera.
Default lib will open with VGA (640x480).
You can ask camera open with another size such as HD, FHD by using
mUVCCameraHellper.setDefaultPreviewSize(1280, 720);
=> pls also try to:
+ get supported preview sizes of camera (you can dig into the UVCCameraHelper.java class to find it)
+ try to open camera with any size there to see which size might be ok to use.
----
Those comments are for testing, I am not sure what your problem is (the gray video). You should analyze the issue by testings.

quocnhat7, 2020-10-15:
"Default lib will open with VGA (640x480)."
=> a faster way: pls open UVCCameraHelper.java, find variable previewWidth, previewHeight and change it to another number.
Such as 1920x1080, 1280x720, 960x640, 640x360, 352x288, ...

heaveninfosoft, 2020-10-15:
right now i m testing on default VGA (640x480)
with i have 3 camera  480p ,720p and 1080p 
issue occur only in 1080p camera will test according to your guide.

i have one more question if really help me than it s good for me
i am doing streaming to youtube ,fb using this project it work well but in  samsung 9 plus(Android 10) with 1080p camera audio recording stop can you guide me to overcome this??

quocnhat7, 2020-10-15:
You can add callback into audio recorder to get exact error code, then trace the reason on google. Perhaps android10 disallow to record in background, maybe. Or after some minutes, Doze mode of phone will disallow app use system resource such as microphone , ... => check error code to confirm this.
----
https://gist.github.com/Venryx/e1f772b4c05b2da08e118ccd5cc162ff
You can ggle some samples such as this.

idr-pt, 2020-10-22:
Gray Video Effect is mocking from Days . Cant fix it . Did everything mentioned

heaveninfosoft, 2020-10-22:
@has yuo got rid from grey line???????

idr-pt, 2020-10-22:
> @Has yuo got rid from grey line???????

NO. Only when Frame format Changed to YUYV . But i need MJPEG

heaveninfosoft, 2020-10-28:
@quocnhat7 in some phone video is blue/violent i can confirm this because of your video encoder can you change so it can work in all mobile??

RayCaiNZ, 2020-10-30:
@jiangdongguo 我也有这个问题灰屏

heaveninfosoft, 2020-10-30:
@RayCaiNZ 

mCameraHelper?.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_MJPEG)
mCameraHelper?.initUSBMonitor(this, mUVCCameraView, listener)
this will fix

@RayCaiNZ i know you we both meet in upwork please msg me so we can work together or at least tell me soution that i can increase project optimization.  some time in dual camera it attach-detach automically please solve me this

idr-pt, 2020-10-30:
> @RayCaiNZ
> 
> mCameraHelper = UVCCameraHelper.getInstance();
> mCameraHelper.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_YUYV);
> this will fix
> 

are you able to capture pics ?

RayCaiNZ, 2020-11-02:
@idr-pt 

Still going grey

JinChangSheng, 2021-05-06:
@jiangdongguo 我也遇到灰屏问题，重新release再打开摄像头可以恢复正常，但是过一会可能又会出现

JinChangSheng, 2021-05-06:
一般设备开机运行时间久了容易出现，预览一段时间，屏幕变成灰色，顶部出现一点画面，预览返回帧率变低

hjywyj, 2024-04-28:
> 一般设备开机运行时间久了容易出现，预览一段时间，屏幕变成灰色，顶部出现一点画面，预览返回帧率变低

我也遇到了，请问解决了吗

zhancaihai, 2024-04-28:
这是来自QQ邮箱的假期自动回复邮件。

您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

eetteri, 2024-08-30:
Reviving old thread as this still seems to be problem (this [issue](https://github.com/jiangdongguo/AndroidUSBCamera/issues/565) opened not so long ago). I don't have clear info what's going on but as I stumbled to this problem on my proof of concept case I did find that the camera image settings can produce this problem.

[Here](https://www.youtube.com/shorts/B2YVwtlO5fk) is short video where I crank gain/saturation/sharpness settings. It almost seems that when camera settings start causing some values in image that are not within some boundaries the frames starts to be corrupted and the grey appears. 

Anybody more knowledgable on the inner workings  of ausbc propably could weight in if this is something related to ausbc library.