jiangdongguo/AndroidUSBCamera #187: 没有预览画面，采集不到数据，调用nativeConnect返回值：-50。

GQii, 2019-11-13:
2019-11-12 17:34:53.215 7669-7730/com.jiangdg.usbcamera E/libuvc/device: [7730*device.c:944:uvc_scan_control]:UVC_ERROR_INVALID_DEVICE 2019-11-12 17:34:53.215 7669-7730/com.jiangdg.usbcamera E/UVCCamera: [7730*UVCCamera.cpp:178:connect]:could not open camera:err=-50 2019-11-12 17:34:53.216 7669-7730/com.jiangdg.usbcamera E/Debug: onError:open failed:result=-50----->id_camera=510743343104;venderId=6610;productId=409;fileDescriptor=60;busNum=1;devAddr=31;usbfs=/dev/bus/usb Exception：调用nativeConnect返回值：-50

SunnyJack, 2019-11-16:
我也遇到同样问题  同问  我是Android 4.4  
11-16 20:43:02.270 7893-7911/com.jiangdg.usbcamera E/libuvc/device: [7911*device.c:944:uvc_scan_control]:UVC_ERROR_INVALID_DEVICE
11-16 20:43:02.270 7893-7911/com.jiangdg.usbcamera E/UVCCamera: [7911*UVCCamera.cpp:178:connect]:could not open camera:err=-50

GQii, 2019-11-17:
哥们，如果解决了，麻烦告诉一声，谢谢。

SunnyJack, 2019-11-17:
我在描述下的我现象（嵌入式设备 Android 4.4），提前插好摄像头 开机 打开应用 %100 会报 err=-50，如果开机后再 插入摄像头 然后在打开 应用 就能够正常预览图像，这个问题困扰我好多天了，一直得不到解决，希望博主 能提供下修改思路 万分感谢！！！

TronQuick, 2019-12-05:
我也是

chengsiyang0814, 2020-04-09:
open failed:result=-50,你们解决了吗

SunnyJack, 2020-04-09:
并没有 ，但是跟摄像头有关系，不是所有摄像头 都这样。

smiledjy, 2020-04-21:
我之前也报这个错，但是打印了下设备发现实usb口占用导致的，源码里 mCameraHelper.requestPermission(0)默认打开的是第一个usb设备，所以报了-50，不知道你们的是什么样的情况，可以把usb设备全打印下看看。

whuan1024, 2020-05-21:
> 我之前也报这个错，但是打印了下设备发现实usb口占用导致的，源码里 mCameraHelper.requestPermission(0)默认打开的是第一个usb设备，所以报了-50，不知道你们的是什么样的情况，可以把usb设备全打印下看看。

谢谢 smiledjy 的答案，解决了我的问题。我也是一开始报这个错，并且没有预览画面，然后试了一下requestPermission(1) 和 requestPermission(2) 等等，就ok了。所以这个和usb节点编号有关。

fragerardi, 2021-02-04:
I get this error when I connect an EasyCap adapter.
Is there any way to fix and view the video like this?

xiaoyanlin1124, 2022-04-16:
1. 感谢@[bbbkkp](https://github.com/bbbkkp)  但是这样并不完美！     Android主板usb接口可能扩展多个usb，mCameraHelper.requestPermission(0); 默认是第一个usb，有时候有可能有，可能无。 细节应该去判断UsbDevice类的设备类型和供应商id，这样比较稳妥。
 // request open permission
            if (!isRequest) {
                if (device != null) {
                    //供应商id
                    int[] productIds = {259, 2825};
                    List<Integer> list = Arrays.asList(ArrayUtils.toObject(productIds));
                    int pro = device.getProductId();
                    boolean isc = list.contains(pro);
                    //设备类型device.getDeviceClass()  
                    if (device.getDeviceClass() == 239 && isc) {
                        mCameraHelper.requestPermission(device);
                        isRequest = true;
                    }
                }
               /* isRequest = true;
                if (mCameraHelper != null) {
                    mCameraHelper.requestPermission(0);
                }*/
            }

xiaoyanlin1124, 2022-04-16:
感谢@[bbbkkp](https://github.com/bbbkkp) 但是这样并不完美！ Android主板usb接口可能扩展多个usb，mCameraHelper.requestPermission(0); 默认是第一个usb，有时候有可能有，可能无。 细节应该去判断UsbDevice类的设备类型和供应商id，这样比较稳妥。
`// request open permission
if (!isRequest) {
if (device != null) {
//供应商id
int[] productIds = {259, 2825};
List list = Arrays.asList(ArrayUtils.toObject(productIds));
int pro = device.getProductId();
boolean isc = list.contains(pro);
//设备类型device.getDeviceClass()
if (device.getDeviceClass() == 239 && isc) {
mCameraHelper.requestPermission(device);
isRequest = true;
}
}
/* isRequest = true;
if (mCameraHelper != null) {
mCameraHelper.requestPermission(0);
}*/
}`

xiaoyanlin1124, 2022-04-16:
谢谢@[bbbkkp](https://github.com/bbbkkp)提醒，代码并不完美。原文mCameraHelper.requestPermission（0）默认设置第一个usb设备为监控设备，这样并不完美，因为有可能一个usb接口扩展多个usb接口。我的做法是判断usb设备类型和供应商id` @Override
        public void onAttachDev(UsbDevice device) {
            // request open permission
            if (!isRequest) {
                if (device != null) {
                    //供应商id
                    int[] productIds = {259, 2825};
                    List<Integer> list = Arrays.asList(ArrayUtils.toObject(productIds));
                    int pro = device.getProductId();
                    boolean isc = list.contains(pro);
                    //设备类型device.getDeviceClass()
                    if (device.getDeviceClass() == 239 && isc) {
                        mCameraHelper.requestPermission(device);
                        isRequest = true;
                    }
                }
               /* isRequest = true;
                if (mCameraHelper != null) {
                    mCameraHelper.requestPermission(0);
                }*/
            }
        }`