jiangdongguo/AndroidUSBCamera #191: ERROR: Failed to resolve: com.serenegiant:common:2.12.4

wangpeng1, 2019-12-18:
能否提供一下这个aar

jiangdongguo, 2020-01-15:
保持网络畅通，多同步几次

jiangdongguo, 2020-01-15:
目前已经升级到4.1.1，很容易拉下来

wuyan001, 2020-02-19:
ERROR: Unable to resolve dependency for ':app@debug/compileClasspath': Could not resolve com.serenegiant:common:4.1.1.  这个一直编译通不过

GeorgeBin, 2020-03-04:
各种版本：https://github.com/saki4510t/libcommon/tree/master/repository/com/serenegiant/common
4.1.1版本：https://github.com/saki4510t/libcommon/tree/master/repository/com/serenegiant/common/4.1.1

Clark-caipeiyuan, 2020-04-27:
use the 
maven{url 'http://raw.github.com/saki4510t/libcommon/master/repository/‘}
or
 maven { url 'https://raw.githubusercontent.com/saki4510t/libcommon/master/repository/' }

Hfengxiang, 2020-05-19:
同问，能否提供一个aar？

Hfengxiang, 2020-05-19:
几周前还好好的项目，昨天打开突然就这样了。。。

GeorgeBin, 2020-05-20:
各种版本aar下载页面：https://github.com/saki4510t/libcommon/tree/master/repository/com/serenegiant/common
4.1.1版本aar下载页面：https://github.com/saki4510t/libcommon/tree/master/repository/com/serenegiant/common/4.1.1