jiangdongguo/AndroidUSBCamera #193: capturePicture not working

jessicanevesmachado, 2019-12-20:
Hello my friends, im try using this project, but when i click to take picture, nothing happing.
this metodo not return result.

 mCameraHelper.capturePicture(picPath, new AbstractUVCCameraHandler.OnCaptureListener() {
                    @Override
                    public void onCaptureResult(String path) {
                        Log.i(TAG,"save path：" + path);
                    }
                });

any can help me?

ailala1993929, 2020-01-08:
issue +1 
mCameraHelper.capturePicture(picPath, new AbstractUVCCameraHandler.OnCaptureListener() {
                    @Override
                    public void onCaptureResult(String path) {
                        Log.i(TAG,"save path：" + path);
                        Toast.makeText(USBCameraActivity.this,"path:"+path,Toast.LENGTH_SHORT ).show();
                    }

ailala1993929, 2020-01-08:
找到问题了：
case MSG_CAPTURE_STILL:
		thread.handleCaptureStill((String)msg.obj);
                //thread.handleStillPicture((String) msg.obj);
                break;
这里的函数调用需要调用上面源码注释掉的函数，thread.handleCaptureStill；
还有一个点就是图片保存路径的问题：String picPath = UVCCameraHelper.ROOT_PATH + MyApplication.DIRECTORY_NAME +"/images/"+ System.currentTimeMillis()
                        + UVCCameraHelper.SUFFIX_JPEG;
这里的images还有外面的USBCamera文件夹需要手动去创建，是bug，所以拍照图片未保存到指定路径。

jessicanevesmachado, 2020-01-13:
thanks Friend for help.
this resolved my problem..
but now im with other problem.. see de the code below

mCameraHelper.capturePicture(picPath, new AbstractUVCCameraHandler.OnCaptureListener() {
                @Override
                public void onCaptureResult(String path) {
                    Log.i(TAG,"save path：" + path);

                        File imgFile = new  File(path);
                        if(imgFile.exists()){

                            Bitmap myBitmap = BitmapFactory.decodeFile(imgFile.getAbsolutePath());
                            imgFoto.setImageBitmap(myBitmap);
                        

                        }

                }
            });

In the code **imgFoto.setImageBitmap(myBitmap);** I get the following error

android.view.ViewRootImpl.checkThread(ViewRootImpl.java:9223)
android.view.ViewRootImpl.requestLayout(ViewRootImpl.java:1602)
android.view.View.requestLayout(View.java:24872)
android.view.View.requestLayout(View.java:24872)
android.view.View.requestLayout(View.java:24872)
android.view.View.requestLayout(View.java:24872)
android.view.View.requestLayout(View.java:24872)
android.view.View.requestLayout(View.java:24872)
android.view.View.requestLayout(View.java:24872)
android.view.View.requestLayout(View.java:24872)
android.widget.ImageView.setImageDrawable(ImageView.java:571)
android.widget.ImageView.setImageBitmap(ImageView.java:705)
com.jiangdg.usbcamera.view.fotoActivity$3.onCaptureResult(fotoActivity.java:189)
com.serenegiant.usb.common.AbstractUVCCameraHandler$CameraThread.saveYuv2Jpeg(AbstractUVCCameraHandler.java:848)
com.serenegiant.usb.common.AbstractUVCCameraHandler$CameraThread.access$400(AbstractUVCCameraHandler.java:391)
com.serenegiant.usb.common.AbstractUVCCameraHandler$CameraThread$3$1.run(AbstractUVCCameraHandler.java:813)
java.lang.Thread.run(Thread.java:764)
异常：Only the original thread that created a view hierarchy can touch its views.

How can I solve this problem?

jessicanevesmachado, 2020-01-14:
hello friend :)
I resolved, using this code

runOnUiThread(new Runnable() {

                        @Override
                        public void run() {

                            //showShortMsg("testando");
                            File imgFile = new  File(UVCCameraHelper.ROOT_PATH +"analiseCapilar" + UVCCameraHelper.SUFFIX_JPEG);
                            if(imgFile.exists()){

                                Bitmap myBitmap = BitmapFactory.decodeFile(imgFile.getAbsolutePath());
                                ImageView img = (ImageView) findViewById(R.id.imgFoto);
                                img.setImageBitmap(myBitmap);
                                img.setVisibility(View.VISIBLE);

                            }

                        }
                    });

inside onCaptureResult

closed 2020-01-14 by jessicanevesmachado

blennerSilva, 2023-08-16:
@jessicanevesmachado-zz  is possile for u to post ur full code ? im having the same issue