jiangdongguo/AndroidUSBCamera #194: 接入人脸识别 失败

ThirdPrince, 2019-12-24:
接入虹软人脸识别，识别不出来，接入android 自带的人脸识别也不行，求指导。。。

jiangdongguo, 2020-01-15:
测试虹软识别没问题   
 @Override
    public void onPreviewResult(byte[] data) {
        int width = mWidth;
        int height = mHeight;
        // 开始识别
        Rect[] rects = ((HomeActivity)getActivity()).handleFaceRecognise(data,width,height);
        // 绘制矩形框
        if (mGLSurfaceView != null) {
            CameraFrameData imageData = mImageDataBuffers.poll();
            if (imageData != null) {
                byte[] buffer = imageData.mData;
                System.arraycopy(data, 0, buffer, 0, buffer.length);
                imageData.mParams = rects;
                mGLSurfaceView.requestRender(imageData);
            }
        }
    }

jiangdongguo, 2020-01-15:
public Rect[] handleFaceRecognise(byte[] data, int width, int height) {
        if (mPresenter == null || mContainer.getChildCount() == 0 || mFloatMenu.isOpened()) {
            if (mLayoutAlert.getVisibility() == View.VISIBLE) {
                hideAlertMessage();
            }
            return null;
        }
        // 检测一帧图像，提取所有人脸特征存储到result
        List<AFT_FSDKFace> result = mPresenter.handlePreviewData(data, width, height);
        if (result.isEmpty()) {
            sendMessage(MSG_NO_FACE, null);
        }
        // 绘制人脸检测矩形框,rects不能为空
        Rect[] rects = new Rect[result.size()];
        for (int i = 0; i < result.size(); i++) {
            rects[i] = new Rect(result.get(i).getRect());
        }
        result.clear();
        return rects;
    }

ThirdPrince, 2020-03-09:
你的人脸是哪个版本？