jiangdongguo/AndroidUSBCamera #200: No Image.

cp0418, 2020-01-13:
When I connected with Logitech camera  It's working but when I connected with Synchronized USB2.0 Stereo Camera It doesn't work then I try to change frame format from MJPEG to YUYV but still doesn't work it's connecting but no image.
[image]

jiangdongguo, 2020-01-15:
please try to update to 2.3.2

cp0418, 2020-01-16:
You mean " classpath 'com.android.tools.build:gradle:3.5.2' " right ?

cp0418, 2020-01-16:
> please try to update to 2.3.2

You mean " classpath 'com.android.tools.build:gradle:3.5.2' " right ?

jiangdongguo, 2020-01-16:
no image perhaps because of creating file directory failed.On 2.3.2,I fixed it.

cp0418, 2020-01-17:
I think the problem is my camera.It supports YUY2. What do you think about this?
[image]

jiangdongguo, 2020-01-17:
nothing different. please using mjpeg and update the lib to 2.3.2,and have a try.

cp0418, 2020-01-18:
I tried and there is an error.
> Could not GET 'https://jitpack.io/com/github/jiangdongguo/libutils/2.3.2/libutils-2.3.2.pom'. Received status code 401 from server: Unauthorized
Enable Gradle 'offline mode' and sync project

I really appreciate you. I am very new about it. Thank you so much for your help.

cp0418, 2020-01-20:
Thank you so much for your help.

closed 2020-01-20 by cp0418