jiangdongguo/AndroidUSBCamera #202: make camera recording in a service without preview

imad-123, 2020-01-15:
hi @jiangdongguo 
i had merge https://github.com/saki4510t/UVCCamera/tree/master/usbCameraTest4  with your project to record videos and take photos using service and it worked. but i have a problem on contrast and brightness can you help me please

jiangdongguo, 2020-01-16:
mCameraHelper.setModelValue(UVCCameraHelper.MODE_BRIGHTNESS,progress);
mCameraHelper.setModelValue(UVCCameraHelper.MODE_CONTRAST,progress);
mCameraHelper.getModelValue(UVCCameraHelper.MODE_BRIGHTNESS);
mCameraHelper.getModelValue(UVCCameraHelper.MODE_CONTRAST);

hi, can not worked?

imad-123, 2020-01-16:
thanks for your answer @jiangdongguo  i don't use mCameraHelper anymore i use camerClient from test4

jiangdongguo, 2020-01-17:
It's Okay,and hope it can give you some informations about your problem.

imad-123, 2020-01-18:
thanks @jiangdongguo

imad-123, 2020-02-03:
hi @jiangdongguo i want to add britness and contrast to test4 can you help me to do that please

MichaelKuzmenko, 2020-06-01:
hi @imad-123 
can you help me to make camera recording without preview?