jiangdongguo/AndroidUSBCamera #206: 401 Unauthorized

patrick-nurt, 2020-02-18:
Hi - great work here!

I keep running this error when I'm building:  

13:02:01.136 [QUIET] [colouredOutputLogger] Build file '/Users/patrickpereira/Viggo/meter/taximeter/platforms/android/app/build.gradle' line: 597
13:02:01.136 [QUIET] [colouredOutputLogger] A problem occurred configuring project ':app'.
13:02:01.136 [QUIET] [colouredOutputLogger] Could not GET 'https://jitpack.io/com/github/jiangdongguo/libutils/2.3.3/libutils-2.3.3.pom'. Received status code 401 from server: Unauthorized

Please advice how I can get around the above.

Thanks in advance.

anantshah93, 2020-02-21:
Same error, any solution?
https://jitpack.io/com/github/jiangdongguo/libutils/2.3.2/libutils-2.3.2.pom There is authentication added so it fails to compile library code in project.

konovalov-k, 2020-02-22:
@anantshah93 Can this libutils based on this stuff?https://github.com/saki4510t/libcommon/tree/master/common/src/main/java/com/serenegiant

Crabgore, 2020-03-05:
> Could not resolve com.github.jiangdongguo:libutils:2.3.2.
     Required by:
         project :app > com.github.jiangdongguo:AndroidUSBCamera:2.3.2
      > Could not resolve com.github.jiangdongguo:libutils:2.3.2.
         > Could not get resource 'https://jitpack.io/com/github/jiangdongguo/libutils/2.3.2/libutils-2.3.2.pom'.
            > Could not GET 'https://jitpack.io/com/github/jiangdongguo/libutils/2.3.2/libutils-2.3.2.pom'. Received status code 401 from server: Unauthorized

same problem.
How can we use your wonderful work?)

MichaelKuzmenko, 2020-03-17:
Same problem.
Please help us.

gesdotstar, 2020-03-28:
Same problem. No way to resolve...

rena2019, 2020-03-30:
seems that the latest working/public accessible version is 2.3.1 (2.3.2 and 2.3.3 do not work)
if you do not need latest version you can use (temporary) solution
 implementation 'com.github.jiangdongguo:AndroidUSBCamera:2.3.1'
in your build.gradle file