jiangdongguo/AndroidUSBCamera #207: ERROR: Unable to resolve dependency for ':app@debug/compileClasspath': Could not resolve com.serenegiant:common:4.1.1.

wuyan001, 2020-02-19:
4.1.1一直编译通不过...

GeorgeBin, 2020-03-04:
将
allprojects {
		repositories {
			...
			maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
			maven { url 'https://jitpack.io' }
		}
	}
中的 http://raw.github.com/saki4510t/libcommon/master/repository/
替换为：https://raw.githubusercontent.com/saki4510t/libcommon/master/repository/

因为源文件的路径由：http://raw.github.com变为-->https://raw.githubusercontent.com

jjjjbababy, 2020-04-23:
aar和pom  文件夹 放到c盘对应的路径下？ aar下载下来怎么用的？

wnpllrzodiac, 2020-05-08:
同问。。。。现在只能在米国的vps上编译通过。。

windyziheng, 2020-05-18:
maven {
url 'https://gitee.com/liuchaoya/libcommon/raw/master/repository/'
}

用这个，有大佬把仓库转到gitee了，编译通过速度贼快
传送门：https://gitee.com/liuchaoya/libcommon/

Hfengxiang, 2020-05-19:
你好，我按你的方式添加了码云的maven地址还是不通过，依然报
`Unable to resolve dependency for ':app@debug/compileClasspath': Could not resolve com.serenegiant:common:4.1.1.`
请问是我遗漏了什么么？

windyziheng, 2020-05-19:
> 你好，我按你的方式添加了码云的maven地址还是不通过，依然报
> `Unable to resolve dependency for ':app@debug/compileClasspath': Could not resolve com.serenegiant:common:4.1.1.`
> 请问是我遗漏了什么么？

项目根目录gradle的allprojects下原来是
maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }

你把这个url替换就行了啊

Hfengxiang, 2020-05-19:
感谢，原来是没有把之前的仓库地址删掉导致的