jiangdongguo/AndroidUSBCamera #212: How to decode the AAC byte[] to Android's raw audio format (I guess PCM)?

acrolink, 2020-03-16:
I want to consume an audio stream captured by the USBCamera and send it over to OpenTok's streaming session using `[getAudioBus().writeCaptureData(ByteBuffer data, int numberOfSamples)]`[1].

What I have is something like this:

```
  mCameraHelper.startPusher(params, new AbstractUVCCameraHandler.OnEncodeResultListener() {
   @Override
   public void onEncodeResult(byte[] data, int offset, int length, long timestamp, int type) {
    Log.d("onEncodeResult", String.valueOf(timestamp));

    // type = 0, aac audio stream
    if (type == 0) {
     customAudioDevice.getAudioBus().writeCaptureData(decodeAAC(data), 44100); // I need a decodeAAC(byte[]) method

    }

   }
  });
```

What is needed is a method to decode the `AAC` byte array into raw format to be sent over to the `AudioBus`.

Your help is highly appreciated. Thanks for the great library.

  [1]: https://tokbox.com/developer/sdks/android/reference/com/opentok/android/BaseAudioDevice.AudioBus.html#writeCaptureData-java.nio.ByteBuffer-int-

jiangdongguo, 2020-03-29:
Sorry, I am late. Your way is right.First you should make sure that  the output of aac is ok, and then check your decode code.Perhaps the best way is save aac to file,and check the data format.

acrolink, 2020-03-29:
I simply edited the `AACEncodeConsumer.java` to bypass encoding to `AAC` altogether in the first place. Thank you for sharing this great library.

closed 2020-04-12 by acrolink

Mehdi-android, 2021-11-23:
Hi @acrolink , can you please tell me what did you edited in `AACEncodeConsumer.java` ? If possible can you share that file to my email mehdisunasara@gmail.com

acrolink, 2021-11-23:
@Mehdi-android 

Here is the modified file:
https://gist.github.com/acrolink/f2433e5f0085ad61be81087c08e1a4d2

Mehdi-android, 2021-11-23:
Thanks for the file, Can I play it while previewing?

bandrarom, 2021-12-21:
did you decode the data?

cristiano-linvix, 2023-01-14:
its possbile to transmit to webrtc audio??