jiangdongguo/AndroidUSBCamera #213: 华为android 9.1 camera open failed

pangxiao2013, 2020-03-17:
很好的工作，我使用最新版本的代码后打开程序连接不上摄像头，用了评论里的所有方法都不行，device_xml.txt文件也有，就是连接不上摄像头。但是这个摄像头还是能用的，从网上下载的usb摄像头app是可以识别的，所以我认为不是设备问题，而是代码兼容性问题，请作者帮忙解决一下。

shenhdbj, 2020-03-24:
您好，请问解决了吗？我也遇到类似问题，我有两个UVC相机，在华为Mate20上其中一个可以使用，另外一个不能使用（请求USB权限确认对话框不能弹出）。同样的在小米手机上两个相机都能正常工作。

pangxiao2013, 2020-03-24:
> 您好，请问解决了吗？我也遇到类似问题，我有两个UVC相机，在华为Mate20上其中一个可以使用，另外一个不能使用（请求USB权限确认对话框不能弹出）。同样的在小米手机上两个相机都能正常工作。

把bulid.grade(module app)中的targetSdkVersion改成27试试

acrolink, 2020-03-29:
Does changing `targetSdkVersion` to 27 help?

shenhdbj, 2020-03-30:
谢谢，把bulid.grade(module app)中的targetSdkVersion改成27已经可以了。

acrolink, 2020-03-30:
But I hope the issue will be fixed so targeting a version higher than 27 would work too. Thanks again for the great work.

AlexanderShniperson, 2020-04-03:
@jiangdongguo 
this is does not help i see camera open error in logs `usb_device_new read returned 543 errno 88`
```
2020-04-03 12:55:20.837 25473-25625/com.jiangdg.usbcamera D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
2020-04-03 12:55:20.853 25473-25473/com.jiangdg.usbcamera V/UVCCameraTextureView: getSurface:hasSurface=true
2020-04-03 12:55:20.853 25473-25473/com.jiangdg.usbcamera V/UVCCameraTextureView: getPreviewTexture:
2020-04-03 12:55:20.854 25473-25625/com.jiangdg.usbcamera I/UVCCameraTextureView: RenderThread#updatePreviewSurface:
2020-04-03 12:55:20.854 25473-25625/com.jiangdg.usbcamera V/UVCCameraTextureView: updatePreviewSurface:tex_id=1
2020-04-03 12:55:20.914 25473-25473/com.jiangdg.usbcamera I/ViewRootImpl: jank_removeInvalidNode all the node in jank list is out of time
2020-04-03 12:55:20.919 25473-25473/com.jiangdg.usbcamera W/InputMethodManager: startInputReason = 1
2020-04-03 12:55:21.146 25473-25551/com.jiangdg.usbcamera W/libEGL: EGLNativeWindowType 0x7799173010 disconnect failed
2020-04-03 12:55:21.153 25473-25473/com.jiangdg.usbcamera D/ViewRootImpl[SplashActivity]: surface should not be released
2020-04-03 12:55:21.163 25473-25473/com.jiangdg.usbcamera D/ActivityThread: Remove activity client record, r= ActivityRecord{b682ae0 token=android.os.BinderProxy@a6b3477 {com.jiangdg.usbcamera/com.jiangdg.usbcamera.view.SplashActivity}} token= android.os.BinderProxy@a6b3477
2020-04-03 12:55:21.772 25473-25622/com.jiangdg.usbcamera I/usbhost: usb_device_new /dev/bus/usb/001/002 fd: 68
2020-04-03 12:55:21.772 25473-25622/com.jiangdg.usbcamera I/usbhost: usb_device_new read returned 543 errno 88
2020-04-03 12:55:21.774 25473-25622/com.jiangdg.usbcamera I/usbhost: usb_device_new /dev/bus/usb/001/002 fd: 82
2020-04-03 12:55:21.774 25473-25622/com.jiangdg.usbcamera I/usbhost: usb_device_new read returned 543 errno 88
2020-04-03 12:55:21.781 25473-25622/com.jiangdg.usbcamera D/UsbDeviceConnectionJNI: close
2020-04-03 12:55:21.785 25473-25622/com.jiangdg.usbcamera I/USBMonitor: name=/dev/bus/usb/001/002,desc=68,busnum=1,devnum=2,rawDesc=[B@4fc2f32
2020-04-03 12:55:21.786 25473-25623/com.jiangdg.usbcamera V/CameraThread: handleOpen:
2020-04-03 12:55:21.786 25473-25623/com.jiangdg.usbcamera V/CameraThread: handleClose:
2020-04-03 12:55:21.786 25473-25623/com.jiangdg.usbcamera V/UVCCameraTextureView: setVideoEncoder:
2020-04-03 12:55:21.786 25473-25625/com.jiangdg.usbcamera V/UVCCameraTextureView: RenderThread#setEncoder:encoder=null
2020-04-03 12:55:21.819 25473-25623/com.jiangdg.usbcamera I/usbhost: usb_device_new /dev/bus/usb/001/002 fd: 82
2020-04-03 12:55:21.819 25473-25623/com.jiangdg.usbcamera I/usbhost: usb_device_new read returned 543 errno 88
2020-04-03 12:55:21.825 25473-25623/com.jiangdg.usbcamera I/usbhost: usb_device_new /dev/bus/usb/001/002 fd: 83
2020-04-03 12:55:21.825 25473-25623/com.jiangdg.usbcamera I/usbhost: usb_device_new read returned 543 errno 88
2020-04-03 12:55:21.831 25473-25623/com.jiangdg.usbcamera D/UsbDeviceConnectionJNI: close
2020-04-03 12:55:21.831 25473-25623/com.jiangdg.usbcamera I/libusb/core: [25623*core.c:2085:libusb_init2]:libusb v1.0.19.10903
2020-04-03 12:55:21.832 25473-25623/com.jiangdg.usbcamera I/libusb/usbfs: [25623*android_usbfs.c:504:op_init2]:start up hotplug event handler
2020-04-03 12:55:21.832 25473-25623/com.jiangdg.usbcamera I/libusb/usbfs: [25623*android_usbfs.c:512:op_init2]:call android_scan_devices
2020-04-03 12:55:21.835 25473-25623/com.jiangdg.usbcamera I/AbsUVCCameraHandler: supportedSize:{"formats":[{"index":1,"type":6,"default":1,"size":["2560x960","2560x720","1280x480","640x240"]}]}
```
i have also decompiled [this Android application](https://play.google.com/store/apps/details?id=com.shenyaocn.android.usbcamera) and saw there is lib `liUVCCamera.so` have bigger size then your

acrolink, 2020-04-03:
@AlexanderShniperson 

A suggest replacing `/libusbcamera/src/main/res/xml/device_filter.xml` with this (i.e. remove all Android 9.0 suggested definitions):

```
<usb>
	<usb-device class="239" subclass="2" />	<!-- all device of UVC -->
</usb>
```
This solved an issue I had. Keep us updated.

AlexanderShniperson, 2020-04-03:
@acrolink no it's doesn't work, there is already only this line, also checked other places

acrolink, 2020-04-03:
Use file explorer for connected device in Android Studio to validate where the USB camera gets mounted. Run `adb root` to see all folder structure.

acrolink, 2020-04-03:
What type of camera are you using?

AlexanderShniperson, 2020-04-03:
[used camera](https://aliexpress.ru/item/32839879959.html)
currently try to understand where is it mounted

AlexanderShniperson, 2020-04-03:
also i should note [this Android application](https://play.google.com/store/apps/details?id=com.shenyaocn.android.usbcamera) works fine with this camera and my smartphone

AlexanderShniperson, 2020-04-03:
also i just saw device is detected, but power led doesn't ON when i choose camera

acrolink, 2020-04-03:
try this to find ou supported dimensions:
`camera.supportedSizeList`

It is possible that the default 640x480 dimensions does not work with this specific camera. You have to change them to the ones supported by the camera.

AlexanderShniperson, 2020-04-03:
ok thank you, will try and reply with result

AlexanderShniperson, 2020-04-06:
hi @acrolink you are right it is related to dimensions, changed it to specific to this camera to 640x240 and it's start working, thank you very much!

TenXu, 2020-04-14:
大佬们好，请问这个问题解决了嘛？我当前APP的targetSdkVersion是28，我改成27毫无疑问是没有问题的，但我想保持在28及28以上，在Android10.0系统无法打开预览，连接摄像头成功之后立马走了onCancel

AlexanderShniperson, 2020-04-19:
@TenXu issue should be fixed by my pull request

TenXu, 2020-04-26:
@AlexanderShniperson 你好
[fix issue 213 related to targetSdk 29 and Android 9.0 or greater; #220
](https://github.com/jiangdongguo/AndroidUSBCamera/pull/220)
这个issue我看过，并且也已试过了，关于Android9.0以上的设备，除了要添加Camera权限之外，对摄像头设备也有要求的
[image]
但是我手头上的设备不足，无法验证这个注释是否正确

AlexanderShniperson, 2020-04-26:
@TenXu please use Google Translate I don’t know how to read your language.
I tested and can say that this fix is works with Android 10, i used device Xiaomi Mi9

xuweihui, 2020-04-28:
华为mate20预览不了，提示,camera open failed，换成红米是可以的

lgstud-io, 2020-06-15:
I made a fix based on https://github.com/jiangdongguo/AndroidUSBCamera/pull/220, but my camera is still not recognized on android 9.

my premissions are fine (as my app has requested camera permission in the runtime).

my intent filters to the activity were also added.

when I connect a USB camera (Canyon CNS-CWC6) the  onAttach(UsbDevice device) is never called. Up until I had android 8 on my tablet it was working with no problems.
If I attach an usb storage, onAttach gets called, it even gets connected.

AlexanderShniperson, 2020-06-17:
@lgstud-io 
it may be related to USB driver that came with that repo.
one of my customers told me about similar problem with Samsung Galaxy J4, but i can't help them because customer taken camera back to himself.