jiangdongguo/AndroidUSBCamera #215: How to change the captured frame size (to something other than 640x480)?

acrolink, 2020-03-29:
I have tried to change the frame size by setting `previewWidth` and `previewHeight` (in `UVCCameraHelper` and `UVCCamera`) to `320x240` but the camera fails to open at all.

I use an iFrameCallback.

My question, how to do that? Alternatively, how to resize the captured image `byte []` to 320x240 ?

I use a Logitech C270 webcam. Thank you.

konovalov-k, 2020-03-30:
Hi,    @acrolink.
First of all get supported res of your camera via UvcCamera.getSuportedSizeList.
Code below use default res (640x480) if custom not supported.

`val check =
                camera.supportedSizeList!!
                    .filter { (it.width in (width-1)..(width+1)) && (it.height in (height-1)..(height+1))}
                    .also {
                        it.forEach {
                            Log.d(TAG, "${it.width} x ${it.height}")
                        }
                    }`
`
            //Use default if not supported cutom res
            if (check.isEmpty()) {
                width = UVCCamera.DEFAULT_PREVIEW_WIDTH
                height = UVCCamera.DEFAULT_PREVIEW_HEIGHT
            }`
If you want to rezise you must convert byte[] to Bitmap and rezise via Bitmap methods. 
Don't forget to select PIXEL_FORMAT_RGBX when set Callback.
Or with LibYuv C++ lib. 

Hope, it could help)

acrolink, 2020-03-30:
Thank you @konovalov-k 

Indeed, the code you provided helped me find the correct dissensions to the start the camera with, Now it works,

Side question, do you prefer Korlin over writing in Java, why is that ?

konovalov-k, 2020-03-30:
@acrolink 
I just started to learn from Kotlin. Better coding perfomance. Better and simple syntax, less code scope. Some multithread features. Overall perfomance the same as in java (google IO 2019). e.t.c.
Of course i can read\write in Java, but if we have better tools, why we can't use it?)
P.S. Don't forget to close issue.

closed 2020-03-30 by acrolink