jiangdongguo/AndroidUSBCamera #216: Could not determine the dependencies of task ':app:processDebugResources'.

superxmmdm, 2020-03-31:
Could not determine the dependencies of task ':app:processDebugResources'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve com.github.jiangdongguo:libutils:2.3.2.
     Required by:
         project :app > com.github.jiangdongguo:AndroidUSBCamera:2.3.2
      > Could not resolve com.github.jiangdongguo:libutils:2.3.2.
         > Could not get resource 'http://raw.github.com/saki4510t/libcommon/master/repository/com/github/jiangdongguo/libutils/2.3.2/libutils-2.3.2.pom'.
            > Could not GET 'http://raw.github.com/saki4510t/libcommon/master/repository/com/github/jiangdongguo/libutils/2.3.2/libutils-2.3.2.pom'.
               > Connection reset
      > Could not resolve com.github.jiangdongguo:libutils:2.3.2.
         > Could not get resource 'https://jitpack.io/com/github/jiangdongguo/libutils/2.3.2/libutils-2.3.2.pom'.
            > Could not GET 'https://jitpack.io/com/github/jiangdongguo/libutils/2.3.2/libutils-2.3.2.pom'. Received status code 401 from server: Unauthorized
跪求各位大佬，这个问题怎么解决

sxp-ilinkin, 2020-03-31:
额 我也是一样的问题

sxp-ilinkin, 2020-03-31:
版本改成2.3.1 就好了

superxmmdm, 2020-03-31:
谢谢，解决了

closed 2020-03-31 by superxmmdm